joukos/PaperTTY #1: SENSATIONAL WORK

assadollahi, 2018-08-14:
My Pi Zero W with the same epaper display was also recently featured by hackday:
https://www.youtube.com/watch?v=1h0B2MWSVmQ
I've put the files on thingiverse.

I had the same idea as you, I wanted to display basic info about the system's state on the display, like wifi ssid / ip address and interact with it. I was really dreaming of having a terminal on it.
You made it happen! Can't wait to try it out at home!

BTW, the bigger Waveshare displays don't have partial refresh and it's really complicated to make them fast enough. There's a bigger display incl partial refresh from papirus: 
https://uk.pi-supply.com/products/papirus-epaper-eink-screen-hat-for-raspberry-pi
I'm planning to design a bigger case for the Pi3 and that display, would love to see your terminal on that one, too.
 
Would be great to stay in touch!

joukos, 2018-08-14:
Thanks! I tried to support as many use cases as possible with this minimal implementation, I hope it works for you. If you encounter some missing steps in the installation instructions or other issues, please let me know and I'll update the docs / try to fix them.

I also noticed the lack of partial refresh in the bigger ones, but my understanding was that [Ben Krasnow worked on it](http://benkrasnow.blogspot.com/2017/10/fast-partial-refresh-on-42-e-paper.html) and managed to make it work with the 4.2" at least. I don't know if there will be some unwanted side effects in the long run.

I also found this [cool 6" display](https://www.youtube.com/watch?v=qK6Smxma_-g) that looks awesome but apparently is not currently sold anywhere...

justinkelly, 2018-08-15:
re 6" with partial refresh - at 51s [1] in that video it says its the GDE060BA﻿ [2] - which seems to available for purchase for $40 [3]

[1] https://www.youtube.com/watch?v=u0RXTDB_Wlo&t=51s
[2] http://www.good-display.com/products_detail/productId=211.html
[3] http://www.buy-lcd.com/index.php?route=product/product&path=2897_8370&product_id=57635

justinkelly, 2018-08-15:
would be amazing if this could work larger screens 6-10" range - will be keeping an eye on this project!!

joukos, 2018-08-25:
Closing (since it's not an actual issue) :)

closed 2018-08-25 by joukos