joukos/PaperTTY #2: TabError: inconsistent use of tabs and spaces in indentation

cclauss, 2018-08-14:
Python 3 treats TabError: as syntax errors

flake8 testing of https://github.com/joukos/PaperTTY on Python 3.7.0

$ __flake8 . --count --select=E901,E999,F821,F822,F823 --show-source --statistics__
```
./papertty.py:312:20: E999 TabError: inconsistent use of tabs and spaces in indentation
	    if not noclear:
                   ^
1     E999 TabError: inconsistent use of tabs and spaces in indentation
1
```

joukos, 2018-08-14:
Thanks, fixed in https://github.com/joukos/PaperTTY/commit/502fffd4a74fe41d5de088043556d67de0475bf0 , was due to sloppiness with editing ;)

The code is for Python 2 though, but consistency is good.

closed 2018-08-14 by joukos

cclauss, 2018-08-14:
500 days until Python 2 end of life...

joukos, 2018-08-15:
Reason for Python 2 is mentioned in `README.md` - I would prefer Python 3.

cclauss, 2018-08-15:
The PIL library that __sudo apt-get install python-imaging__ is installing has not been update in almost 10 years.  http://www.pythonware.com/products/pil/ and Pillow is the Python community's supported replacement.

http://pillow.readthedocs.io -- __python-imaging__ should be uninstalled before installing Pillow.__

You might be able to run on either Python 2 or Python 3 by replacing the line:
* __sudo apt-get install python-imaging python click__
with:
* __sudo apt-get remove python-imaging__  # and then
* __sudo python2 -m pip install --upgrade Pillow click__  # and/or
* __sudo python3 -m pip install --upgrade Pillow click__
    * https://pypi.org/project/Pillow
    * https://pypi.org/project/click

Another route would be to replace __sudo apt-get install python-imaging python click__ with:
* __sudo apt-get remove python-imaging__
* __sudo apt-get install python-pil python click__

joukos, 2018-08-15:
The Waveshare library uses `python-imaging` - I didn't want to make changes to their code (others have already done that). Feel free to fork/PR with your changes so it's modern, Python 3 compatible and still works hassle-free on Raspbian Stretch. (and perhaps create another issue)

joukos, 2018-08-15:
I created a new issue (#3) for replacing PIL.