joukos/PaperTTY #3: Replace PIL with Pillow

joukos, 2018-08-15:
This issue is for future reference.

@cclauss outlined some steps to replace the imaging library with Pillow in https://github.com/joukos/PaperTTY/issues/2#issuecomment-413100455.

Currently there's some wrinkles in going for it because the Waveshare library uses PIL. There are forks around already.

cclauss, 2018-08-15:
A pointer to those forks please?

joukos, 2018-08-15:
I probably won't have time to work on this for the rest of the week, but should use the repo above and make the necessary changes to instructions.

cclauss, 2018-08-15:
Yes.  It would be a welcome addition if someday there were Python repos at https://github.com/waveshare

joukos, 2018-08-24:
Fixed in 7e557abdc8a737d8f94e5eee1a07f802a85a572e.

closed 2018-08-24 by joukos