joukos/PaperTTY #9: Cannot start papertty.py, missing PIL error

albill, 2018-08-30:
On a clean June 27, Raspbian Stretch Lite install on a Pi Zero W 1.1, I followed your installation instructions.

I set up the virtualenv with:

mkvirtualenv -p /usr/bin/python3 -r requirements.txt papertty

which says (the second time I ran it to be sure, which is why they're already satisfied):

Installing setuptools, pkg_resources, pip, wheel...done.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: Pillow in /home/pi/.virtualenvs/papertty/lib/python3.5/site-packages (from -r requirements.txt (line 1)) (5.2.0)
Requirement already satisfied: RPi.GPIO in /home/pi/.virtualenvs/papertty/lib/python3.5/site-packages (from -r requirements.txt (line 2)) (0.6.3)
Requirement already satisfied: spidev in /home/pi/.virtualenvs/papertty/lib/python3.5/site-packages (from -r requirements.txt (line 3)) (3.2)
Requirement already satisfied: click in /home/pi/.virtualenvs/papertty/lib/python3.5/site-packages (from -r requirements.txt (line 4)) (6.7)

When I finally get into the PaperTTY virtualenv following the above, I receive the following when I run: 

sudo ./papertty.py list

Traceback (most recent call last):
  File "./papertty.py", line 14, in <module>
    import drivers.drivers_base as drivers_base
  File "/home/pi/PaperTTY/drivers/drivers_base.py", line 18, in <module>
    from PIL import Image
ImportError: No module named 'PIL'

Any suggestions? This is a completely clean Pi Zero and I just imaged it from Raspbian.

joukos, 2018-08-30:
Hmm... do you get similar output from these commands in the virtualenv?

```
(papertty) pi@rasputin:~/code/PaperTTY $ pip list
Package       Version
------------- -------
click         6.7
Pillow        5.2.0
pip           18.0
pkg-resources 0.0.0
RPi.GPIO      0.6.3
setuptools    40.1.0
spidev        3.2
wheel         0.31.1
(papertty) pi@rasputin:~/code/PaperTTY $ pip --version
pip 18.0 from /home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/pip (python 3.5)
(papertty) pi@rasputin:~/code/PaperTTY $ which python; python --version
/home/pi/.virtualenvs/papertty/bin/python
Python 3.5.3
```

albill, 2018-08-30:
My version:

`~/PaperTTY $ pip list
Package       Version
------------- -------
click         6.7    
Pillow        5.2.0  
pip           18.0   
pkg-resources 0.0.0  
RPi.GPIO      0.6.3  
setuptools    40.2.0 
spidev        3.2    
wheel         0.31.1 

pip --version
pip 18.0 from /home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/pip (python 3.5)

:~/PaperTTY $ python --version
Python 3.5.3
`

albill, 2018-08-30:
Seems that's not the only one. After installing the package at the system level and trying again, I get:

```
~/PaperTTY $ sudo ./papertty.py list
Traceback (most recent call last):
  File "./papertty.py", line 14, in <module>
    import drivers.drivers_base as drivers_base
  File "/home/pi/PaperTTY/drivers/drivers_base.py", line 19, in <module>
    import spidev
ImportError: No module named 'spidev'
```
I suspect that it is pulling multiple packages from the system.

joukos, 2018-08-30:
Hmm, in that case it appears the virtualenv doesn't work as intended...

joukos, 2018-08-31:
Ah, ok, this is because of `sudo`, ie. the environment changes.

Two ways to run it:

Activating the virtualenv within the `sudo` command:
```sh
sudo bash -c ". ~pi/.virtualenvs/papertty/bin/activate; ./papertty.py --driver epd2in13 terminal --autofit"
```

Or, simpler, just running the interpreter directly, ensuring the env gets set:
```sh
sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py --driver epd2in13 terminal --autofit
```

I'll update the docs. Thanks for submitting the issue.

joukos, 2018-08-31:
Fixed in https://github.com/joukos/PaperTTY/commit/cea7132bf8b4960be86831a5f39d6920754964d6.

closed 2018-08-31 by joukos