joukos/PaperTTY #10: Fix encoding issues

joukos, 2018-09-02:
Implement "proper" support for different encodings and error handling, meaning that the program shouldn't for example crash because of a character that can't be drawn. For now it just barely works.

There's a few things to consider, such as:

- virtual console charset (and char device to read from)
- locale settings
- glyph support in various fonts
- `tmux` supports UTF-8 only, for example
- etc.

Goal would be to handle any garbage without problems and displaying it as well as possible using default settings - if this were to be used as the only means to get output from a RPi, it should never crash.

jason-green-io, 2018-09-06:
Fixing the encoding issues would be great. It'd be even better if you can get the UTF-8 data and supplement Tom-thumb with some glyphs for line drawing or some of the block glyphs in Unicode. It would allow for printing of qrcodes along side text data within the 20x62 of the 2n13. Super useful for 2fa uses.

joukos, 2018-09-06:
Neat idea with the QR codes. Adding 3 block glyphs doesn't sound like too much work, so it could probably be done with decent font editing skills - that I lack for now :)

jason-green-io, 2018-09-06:
I'll see what I can do

joukos, 2018-09-06:
Oh, one problem with those could be that they might not work when reading the virtual console via `vcsa`... since it doesn't store the original unicode, the glyphs would need to represent a value below 512 (and even then, reading the 9th bit requires an ioctl that the code doesn't do at the moment), so it's a bit tricky.

If pseudo TTY support was added first, this should be easier (it's a feature that's missing anyway). That or waiting for the kernel to have `/dev/vcsu*`...

joukos, 2018-09-06:
Issue for pty here: https://github.com/joukos/PaperTTY/issues/16

joukos, 2018-09-06:
By the way, would `qrencode -t ASCIIi` yield a readable code?

jason-green-io, 2018-09-07:
There aren't enough lines with Tom Thumb to show the full QR code. Needs 21 lines, there are 20 - 1 for tmux. BUT!

This font is one pixel taller but supports more graphical glyphs: https://github.com/josuah/miniwi

Converts well with pilfont.. just need those code points ;)

joukos, 2020-01-22:
Now that the kernel (since 4.19 or so) includes the `vcsu` device, Unicode support is currently being implemented in https://github.com/joukos/PaperTTY/pull/45 - this issue will be closed when it's done.

joukos, 2020-01-22:
https://github.com/joukos/PaperTTY/pull/45 was merged and likely fixes this nicely, so closing this issue.

closed 2020-01-22 by joukos