joukos/PaperTTY #12: Flashing?

wle8300, 2018-09-02:
Hi there! General question:

Does the entire screen flash whenever something changes?

joukos, 2018-09-02:
Not necessarily, this depends on whether the display supports partial refresh. As of now, the supported displays that officially handle partial refresh are:

- EPD1in54 - EPD 1.54" (monochrome)
- EPD2in13 - EPD 2.13" (monochrome)
- EPD2in13d - EPD 2.13" D (monochrome, flexible)
- EPD2in9 - EPD 2.9" (monochrome)

With these models, just the updated part will change, without flickering. This may be just the area of a single character, or even the entire screen (check the Youtube videos for examples).

Do note that I only own the `EPD2in13` model myself, so it's the only one I've tested this with, but it is my *guess* that all of the above behave similarly, judging by the similarity of the driving code. With the 2.13" model there is no flickering and it works quite nicely, ie. responsive enough for IRC, system administration etc.

The 4.2" monochrome display has been shown to manage partial refresh with [LUTs created by Ben Krasnow](https://hackaday.com/2017/10/31/ben-krasnow-hacks-e-paper-for-fastest-refresh-rate/) but I can't really add proper support for that until I get the hardware to test it with.

With the other displays - *for now* - whenever the screen is updated a full refresh is made, which means some flashing will likely occur. You can adjust the sleep delay to make it more bearable for interactive use, but the displays that don't support partial refresh are probably more useful for stuff that doesn't update too often (ie. once a minute or so). The displays with an extra color - red/yellow - are supposedly quite slow to update, although PaperTTY itself does not use the color frames for now.

That said, I've only tried this with the 2.13" display, so I can't really say much for the other models. They might simply not work at all, but in that case I'm very interested to get feedback so I can fix it.

If you intend to buy one to experiment with interactive use, I'd recommend either the ones with official partial refresh support. Or - **if you're brave** - the 4.2" one, as it *may* get partial refresh support later. Still, **no guarantees**.

joukos, 2018-09-02:
Closing for now.

closed 2018-09-02 by joukos