joukos/PaperTTY #14: 4.2 inch display doesn't work

albill, 2018-09-03:
I have a Pi 3 B+ system where I've been successfully using a 2.9 inch display. I ordered a monochrome 4.2 inch display, which arrived today. I connected it up and changed it from epd2in9 to epd4in2 in start.sh so I would get console on it on boot but nothing happened (not even the scrub). 

My start.sh's command line is:

`${VENV} papertty.py --driver epd4in2 terminal --font /usr/share/fonts/truetype/msttcorefonts/Andale_Mono.ttf --size 12 --scrub --autofit`

I investigated by stopping the service and then trying to run a scrub by hand with:

`sudo ./papertty.py --driver epd4in2 scrub`

and I get the following error:

```
~/PaperTTY $ sudo ./papertty.py --driver epd4in2 scrubTraceback (most recent call last):
  File "./papertty.py", line 443, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "./papertty.py", line 310, in scrub
    ptty.driver.scrub(fillsize=size)
  File "/home/pi/PaperTTY/drivers/drivers_base.py", line 53, in scrub
    self.fill(self.black, fillsize=fillsize)
  File "/home/pi/PaperTTY/drivers/drivers_base.py", line 60, in fill
    self.draw(x, 0, image)
  File "/home/pi/PaperTTY/drivers/drivers_full.py", line 66, in draw
    self.display_frame(self.get_frame_buffer(image))
  File "/home/pi/PaperTTY/drivers/drivers_full.py", line 76, in get_frame_buffer
    ({0}x{1}).'.format(self.width, self.height))
ValueError: Image must be same dimensions as display                 (400x300).
```
I don't think I am doing anything obviously wrong and this worked on the 2.9 on the same Raspberry Pi.

joukos, 2018-09-03:
Nice to know it worked on the 2.9" :) Thanks for reporting. I'll try to take a look tomorrow. Perhaps add a statement to print the supposed image dimensions (`print(imwidth, imheight)`) on line 75 of `papertty.py` to see what they are set to, might be simple to fix.

albill, 2018-09-04:
Nethack should fit on one a lot better than a 2.9". :-)

joukos, 2018-09-04:
Does `--portrait` do something?

albill, 2018-09-04:
```
sudo ./papertty.py --driver epd4in2 terminal --portrait
Started displaying /dev/vcsa1, minimum update interval 0.1 s, exit with Ctrl-C
```

and then I get a console visible!

joukos, 2018-09-04:
Neat :) This is something I was going to fix/simplify but didn't get around to it yet, ie. since the screens have varying dimensions and ways of going through the pixels, the current code probably just breaks because of the rotation that is done in "landscape" mode. The 2.13" is "naturally" a portrait screen (Y axis is longer), but with the 4.2" it's the other way around.

Should fix it so that the setting takes that into account - or, just get rid of the confusing term altogether and have a rotation setting instead, which sounds more useful.

albill, 2018-09-04:
I'll note that while --portrait works alone, adding it to my sequence for a terminal console on boot still errors. I tried running what I have in start.sh by hand:

```
(papertty) pi@raspberrypi:~/PaperTTY $ ./papertty.py --driver epd4in2 terminal --font /usr/share/fonts/truetype/msttcorefonts/Andale_Mono.ttf --size 14 --scrub --portrait
Traceback (most recent call last):
  File "./papertty.py", line 443, in <module>
    cli()
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/decorators.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "./papertty.py", line 363, in terminal
    ptty.driver.scrub()
  File "/home/pi/PaperTTY/drivers/drivers_base.py", line 53, in scrub
    self.fill(self.black, fillsize=fillsize)
  File "/home/pi/PaperTTY/drivers/drivers_base.py", line 60, in fill
    self.draw(x, 0, image)
  File "/home/pi/PaperTTY/drivers/drivers_full.py", line 66, in draw
    self.display_frame(self.get_frame_buffer(image))
  File "/home/pi/PaperTTY/drivers/drivers_full.py", line 76, in get_frame_buffer
    ({0}x{1}).'.format(self.width, self.height))
ValueError: Image must be same dimensions as display                 (400x300).
```

albill, 2018-09-04:
(I normally had --autofit in there but I removed it for troubleshooting)

joukos, 2018-09-04:
Hmm, is it the scrubbing that doesn't work but terminal does?

albill, 2018-09-04:
Yes, if I remove --scrub, suddenly the console appears with that command line though I had to use sudo as I got this before I did (on the same machine where I didn't need sudo five minutes ago):

```
./papertty.py --driver epd4in2 terminal --font /usr/share/fonts/truetype/msttcorefonts/Andale_Mono.ttf --size 14 --portrait
No read access to /dev/vcsa1 - maybe run with sudo?
```

joukos, 2018-09-04:
Normally `sudo` should be needed. To fix the bugs, I think I need to take a look if there's any fiddling required with adding arbitrary rotation instead of just two orientations. It ought to fix such issues.

albill, 2018-09-04:
Strangely, if I boot with this in my start.sh, nothing visible happens but if I stop and start papertty with systemctl manually from ssh'ing in, suddenly the screen shows the console. Something is broken on boot though even though both use the same start.sh.

```
VENV="/home/pi/.virtualenvs/papertty/bin/python3"
${VENV} papertty.py --driver epd4in2 terminal --font /usr/share/fonts/truetype/msttcorefonts/Andale_Mono.ttf --size 12 --autofit --portrait
```

joukos, 2018-09-04:
Have you checked with `journalctl -u papertty` if there are exceptions in the log during boot?

Although this could also be similar to what I accidentally had happen before: if the colors are inverted by some faulty logic and the service is stopped, it attempts to clear the screen - but with black instead of white, appearing as if it didn't draw anything before. You could try adding `--noclear` and see if the screen has any content after stopping the program, or if it makes any difference at all.

joukos, 2018-09-04:
Hmm, now that I look at it, I get the feeling I got it wrong here: https://github.com/joukos/PaperTTY/blob/master/drivers/drivers_full.py#L68

Try changing that `reverse = False` to `reverse = True`. I think I got it the wrong way there :)

albill, 2018-09-05:
journalctl -u papertty shows me:

```
Sep 04 21:54:46 raspberrypi start.sh[88]: Automatic resize of TTY to 28 rows, 56 columns
Sep 04 21:54:46 raspberrypi start.sh[88]: TTY refused to resize (rows=28, cols=56), continuing anyway.
Sep 04 21:54:46 raspberrypi start.sh[88]: Try setting a sane size manually.
Sep 04 21:54:46 raspberrypi start.sh[88]: Started displaying /dev/vcsa1, minimum update interval 0.1 s, exit with Ctrl-C
Sep 04 21:54:46 raspberrypi start.sh[88]: Traceback (most recent call last):
Sep 04 21:54:46 raspberrypi start.sh[88]:   File "papertty.py", line 443, in <module>
Sep 04 21:54:46 raspberrypi start.sh[88]:     cli()
Sep 04 21:54:46 raspberrypi start.sh[88]:   File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 7
Sep 04 21:54:46 raspberrypi start.sh[88]:     return self.main(*args, **kwargs)
Sep 04 21:54:46 raspberrypi start.sh[88]:   File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 6
Sep 04 21:54:46 raspberrypi start.sh[88]:     rv = self.invoke(ctx)
Sep 04 21:54:46 raspberrypi start.sh[88]:   File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 1
Sep 04 21:54:46 raspberrypi start.sh[88]:     return _process_result(sub_ctx.command.invoke(sub_ctx))
Sep 04 21:54:46 raspberrypi start.sh[88]:   File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 8
Sep 04 21:54:46 raspberrypi start.sh[88]:     return ctx.invoke(self.callback, **ctx.params)
Sep 04 21:54:46 raspberrypi start.sh[88]:   File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 5
Sep 04 21:54:46 raspberrypi start.sh[88]:     return callback(*args, **kwargs)
Sep 04 21:54:46 raspberrypi start.sh[88]:   File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/decorators.py", 
Sep 04 21:54:46 raspberrypi start.sh[88]:     return f(get_current_context().obj, *args, **kwargs)
Sep 04 21:54:46 raspberrypi start.sh[88]:   File "papertty.py", line 423, in terminal
Sep 04 21:54:46 raspberrypi start.sh[88]:     buff = ''.join([r.decode() + '\n' for r in ptty.split(buff, cols)])
Sep 04 21:54:46 raspberrypi start.sh[88]:   File "papertty.py", line 423, in <listcomp>
Sep 04 21:54:46 raspberrypi start.sh[88]:     buff = ''.join([r.decode() + '\n' for r in ptty.split(buff, cols)])
Sep 04 21:54:46 raspberrypi start.sh[88]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xfe in position 45: invalid start by
Sep 04 21:54:46 raspberrypi systemd[1]: papertty.service: Main process exited, code=exited, status=1/FAILURE
Sep 04 21:54:46 raspberrypi systemd[1]: papertty.service: Unit entered failed state.
Sep 04 21:54:46 raspberrypi systemd[1]: papertty.service: Failed with result 'exit-code'.
Sep 04 21:54:47 raspberrypi systemd[1]: papertty.service: Service hold-off time over, scheduling restart.
Sep 04 21:54:47 raspberrypi systemd[1]: Stopped PaperTTY.
```

albill, 2018-09-05:
> Hmm, now that I look at it, I get the feeling I got it wrong here: https://github.com/joukos/PaperTTY/blob/master/drivers/drivers_full.py#L68
>
> Try changing that reverse = False to reverse = True. The original comments in display_frame indicate that # bit set: white, bit reset: black but for some reason I'm doing the opposite :)

This did nothing.

albill, 2018-09-05:
Interestingly, when I try this now, I get the following:

```
sudo ./papertty.py --driver epd4in2 terminal --portrait
Started displaying /dev/vcsa1, minimum update interval 0.1 s, exit with Ctrl-C
Traceback (most recent call last):
  File "./papertty.py", line 443, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "./papertty.py", line 423, in terminal
    buff = ''.join([r.decode() + '\n' for r in ptty.split(buff, cols)])
  File "./papertty.py", line 423, in <listcomp>
    buff = ''.join([r.decode() + '\n' for r in ptty.split(buff, cols)])
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xfe in position 28: invalid start byte
```
Is it disliking the rendering of some character in the boot console?

joukos, 2018-09-05:
That error is related to https://github.com/joukos/PaperTTY/issues/10

You don't seem to have the latest changes, I added an `--encoding` switch earlier and fixed it a bit, so you may want to do a pull. It can also crash if the font doesn't have the character, in which case trying another font may help.

albill, 2018-09-05:
After doing a git pull, I no longer get the error message and get a console on boot (with 'terminal').

On to the next 4.2 problem: whenever it refreshes at all, the whole screen blanks white, then black, then white, then black again before re-rendering the existing text. It is doing a full screen wipe (twice?) any time it refreshes. So hitting enter or space on the control triggers this with every character change. I could post a video but I suspect you know what this looks like.

joukos, 2018-09-05:
Yeah, with full refresh it does flash a few times. Good news is that I got my own 4.2" in the mail today, so I can now try implementing partial refresh for it and fix the bugs. I already worked a bit on the rotation so it will probably work better in the next update. It'll take some time, though.

albill, 2018-09-05:
So there is no way to do partial refresh at this point on the 4.2?

joukos, 2018-09-05:
Not unless you integrate the "Krasnowian" LUT for it yourself :) I briefly tried, but didn't get it to work just yet. The code is for Arduino with limited memory, so a simple copy-paste is not quite enough.

joukos, 2018-09-05:
I created a new issue for the partial refresh (https://github.com/joukos/PaperTTY/issues/15) and will close this one for now, as the display (sort of) works.

closed 2018-09-05 by joukos