joukos/PaperTTY #15: Implement partial refresh for 4.2" module

joukos, 2018-09-05:
There's plenty of tweaking to do, but the 4.2" module seems to more or less work with full refreshes. Now the partial refresh should be implemented for it.

joukos, 2018-09-05:
[Relevant files and info](http://benkrasnow.blogspot.com/2017/10/fast-partial-refresh-on-42-e-paper.html)

joukos, 2018-09-05:
Success! Kind of. Needs some code changes here and there, but the display actually does do partial refreshes. Hopefully with some tweaks it works well enough to be usable, we'll see.

joukos, 2018-09-09:
This has been on hold for a while, since I encountered some weird behavior with the display that I'll try to make sense of first before going further. The 4.2" module is fortunately very easy to hook up to a logic analyzer...

joukos, 2018-09-11:
Display seems to work again... it was in a state where it was constantly "busy" after power on, so I added a generous timeout to `wait_until_idle` and adjusted `VCM_DC_SETTING` because the image was randomly more or less washed out. After updating the image a few dozen times (and waiting a couple of days), it now seems to work as it should again.

It sounds like someone else might have had [something similar](https://forum.arduino.cc/index.php?topic=487007.msg3611910#msg3611910) happen with their 7.5" in an unrelated scenario.

Anyway, now that that's out of the way, I can perhaps look at the partial updates at some point, although I'm a bit busy right now.

f0nk, 2018-11-16:
Hi!
Did you ever have any luck here?

I’m working on a similar project with the 4.2.

Br

joukos, 2018-11-17:
The partial refresh seemed to work, but the code needs some fixes here and there so I haven't had the chance to really test it yet. Support for "proper" rotation is one thing that should be implemented to avoid problems like the code attempting to draw outside the display (which might put it into a weird state). I worked on it a bit but it's currently in a very unfinished state.

I'd like to incorporate the pseudo TTY functionality too at the same time to make the whole thing so much more useful and the code more straightforward, but I simply don't have the time right now to work on this project. :disappointed:

f0nk, 2018-11-17:
Hi Joukos,
Thanks for your quick reply. No problem - I will let you know if I have any luck :)

Br

joukos, 2019-09-16:
@NullandKale I'll see what I can do. I'd need to dig the source mess a bit first to remember what state I left it in, but I'm not sure yet when I'll have the spare time to do that. AFAIR there wasn't much extra code to get the partial refresh somewhat working.

joukos, 2020-03-01:
Now implemented with https://github.com/joukos/PaperTTY/pull/40, so closing this.

closed 2020-03-01 by joukos