joukos/PaperTTY #16: Implement pseudo-terminal

joukos, 2018-09-06:
Instead of reading `/dev/vcsa*`, allocate a pseudo TTY.

A big problem with the VT buffer is that the original unicode value is lost.

Having an option to use a pty should:
- make the encoding issues easier to handle
- enables the use of more symbols, block characters, languages etc.
- allow to have an arbitrarily sized terminal, instead of ioctl complaining - currently some of the display/font combinations are such that not all of the panel can be utilized optimally

jason-green-io, 2018-09-07:
This should help

https://github.com/selectel/pyte

joukos, 2018-09-07:
Thanks, I better check that out.

jason-green-io, 2018-09-09:
I've been working on this with some pretty good success, I'll create a branch on my fork later so you can see.

[image]
[image]

joukos, 2018-09-09:
Looks awesome, nice work!

jason-green-io, 2018-09-09:
https://github.com/jason-green-io/PaperTTY/tree/pyte

jason-green-io, 2018-09-09:
Wish I saw this earlier

https://github.com/pielgrzym/paper-terminal

joukos, 2018-09-09:
:astonished: Hidden in plain sight it seems! Cool find, also an interesting way to show the tty1, didn't occur to me to do it like that.

We should combine all the useful code of various such projects into a package that not only has all the features one might need, but also comes with good drivers (or at least less hacky demo code). From what I can gather via forums and googling, the [GxEPD](https://github.com/ZinggJM/GxEPD) might have useful bits too especially in terms of driving the displays.

jason-green-io, 2018-10-15:
Something else pseudo-terminal enables: attributes

[image]

joukos, 2018-10-15:
Cool :) I hope I have time in the near future to incorporate this and fix some pending driver issues too.

klundry, 2018-10-24:
This is looking really promising! I will be happy to test this out once it's implemented.

lvan188, 2020-10-24:
How to enter the pseudo terminal I have opened

joukos, 2020-10-25:
> How to enter the pseudo terminal I have opened

Are you using the fork by @jason-green-io ? I think you can do a `tmux attach -t tty` to enter the session but haven't tried it myself.