joukos/PaperTTY #18: display terminal dashboards blessed-contrib (ascii/ansi) not working

BenRoe, 2018-09-23:
Hello,
i try to use [blessed-contrib](https://github.com/yaronn/blessed-contrib) with you PaperTTY script.

I start your script with...
`sudo ~/PaperTTY/papertty.py --driver EPD2in9 terminal --sleep 2 --autofit`

open another terminal window
`sudo openvt -fc 1 -- node ./examples/map.js`

Only the x is displayed.
I tried to change the color and made it monochrome, but no luck.

Any ideas?

joukos, 2018-09-23:
Seems it uses the braille glyphs - the default font doesn't have these, and because they are higher up in the unicode range they won't really work with the VT either (because the original unicode value is not stored in the buffer). One "option" might be to make the map uglier and modify the code that draws it to use some more common characters.

For the time being, you could look into what @jason-green-io did here: https://github.com/joukos/PaperTTY/issues/16, that is, a simple pseudo terminal implementation so you can use more complex fonts.

I'm currently too busy IRL to work on it, need to move and stuff.

P.S. that looks like a nifty piece of software :)

BenRoe, 2018-09-23:
Thank you for the quick answer. Try to build a dashboard with a e-paper display, but it's not easy to find a solution without writing a complete python script to draw the stuff on the display.
Would be great to use a "framework" like [blessed-contrib](url
https://github.com/yaronn/blessed-contrib), or [MagicMirror](https://github.com/MichMich/MagicMirror). 
Other idea could be, using MagicMirror and make a screenshot with a python script (https://stackoverflow.com/a/12031316/470707) and draw it `ImageDraw.Draw` to the display periodically.

joukos, 2018-09-24:
It should (probably) work after https://github.com/joukos/PaperTTY/issues/16 is done.

Another thing I'd like to try if I ever have enough time is implementing `PaperVNC`, which should allow running graphical programs and make stuff easier still. In a certain way it would be "simpler" to implement than messing with terminal specifics.

Anyway, I'll close this issue for now, as the root cause already has an issue.

closed 2018-09-24 by joukos