joukos/PaperTTY #19: Scrub command error with EPD 2.7" (monochrome)

mussabo, 2018-10-15:
Hi, thank you very much for the code, this is really great work.
It's my first post in Github in my life, but it seems I've found error, so just made account of Github.
I'm not good at programming, so posting this with hoping this is right manner...

I'm using  EPD 2.7" (monochrome) of waveshare,
and tried command 
sudo ./papertty.py --driver epd2in7 scrub

but got error (text file attached) hope this can be useful.
[error.txt](https://github.com/joukos/PaperTTY/files/2479233/error.txt)

by the way I have EPD1.54" also, and command worked for this display.

joukos, 2018-10-15:
Does it work if you pass `--portrait`?

mussabo, 2018-10-15:
ok I'll try now

mussabo, 2018-10-15:
Is this the right command ??

sudo ./papertty.py --driver epd2in7 scrub --portrait

joukos, 2018-10-15:
Oh, right, the scrub command itself doesn't have that argument. With a full refresh screen you perhaps shouldn't even need the scrub command. Try it with the terminal command.

```sh
sudo ./papertty.py --driver epd2in7 terminal --portrait
```

joukos, 2018-10-15:
These displays are still work-in-progress since I only fairly recently got one (the 4.2") - I have some unfinished code changes that should make them work better (and handle rotation etc.) when I have time to merge them, but I've been very busy for the last few weeks.

mussabo, 2018-10-15:
terminal screen seems working, thank you for the reply !

joukos, 2018-10-15:
Great, I'll close this issue for now.

closed 2018-10-15 by joukos