joukos/PaperTTY #20: CPU usage

klundry, 2018-10-19:
What is the expected CPU load on a Pi Zero? I'm seeing constant 50-70% load on my Zero just sitting at a terminal prompt and I'm curious if that is expected for this hardware or something is wonky.

joukos, 2018-10-19:
It really shouldn't use much (~1% or so depending on the sleep delay) unless the screen is being refreshed. In that case either the screen buffer content was changed or the cursor moved.

Maybe add a debug print statement to somewhere around https://github.com/joukos/PaperTTY/blob/master/papertty.py#L429 to see if it thinks something changed?

klundry, 2018-10-20:
Ah, ok I found the issue. I was checking the cpu load with top and htop on the display itself. Top and htop have so much constantly changing information that it was keeping the cpu load pretty high. When I leave it at a prompt and check top via ssh on another computer it stays around 2-3% for papertty and 5% overall load.

joukos, 2018-10-21:
That would explain it :) I'll close this now.

closed 2018-10-21 by joukos