joukos/PaperTTY #24: Multiple screen support?

kodonnell, 2018-11-03:
Would this be in scope for this project? Hardware issues aside, I imagine straightforward configurations (e.g. 2x2 screens of the same size) would be pretty trivial to code. Of course, hardware would be the main problem and this isn't my area - does anyone have any ideas, or can link relevant discussions? Naively, it looks like only 8 pins are needed, and the Pi has more than that ... so maybe?

kodonnell, 2018-11-03:
[This](https://forum.arduino.cc/index.php?topic=487007.msg3492271#msg3492271) shows it's possible on an Arduino ...

joukos, 2018-11-03:
Yes, I've considered it and there shouldn't be big obstacles in implementing it, other than my lack of time to do it (and limited pool of hardware).

This should be taken into consideration also when eventually implementing the pseudo TTY code so that the image could be stretched to fill several screens, or to have many terminals.

kodonnell, 2018-11-03:
> there shouldn't be big obstacles in implementing it

Great! I'd be interested in trying this, but given I've got no hardware and minimal understanding of how this works ... don't rely on me.