joukos/PaperTTY #25: Support waveshare 9.7in display

phaer, 2018-11-03:
Hi,

I acquired an [9.7 inch e-ink display](https://www.waveshare.com/9.7inch-e-Paper-HAT.htm) which, according to the waveshare, supports partial refresh. I've used their example code for full refreshes, but did not yet find out how to do a partial refresh - the displays refreshes partially, but doesn't display new text in the refreshed area - yet ;)

I'd be very interesting in working on PaperTTY support for this display, but I am new to e-ink displays in general and the 9.7in model seems to work quite differently compared to the smaller models. It uses a IT8951 controller. As far as I understand, I'd need the LUT values for PaperTTY? They seem to be stored in the hat, at least I haven't found them in the demo code yet. And information on the public internet seems sparse.

Anyone else interested in this?

More Infos on the official Wiki page: https://www.waveshare.com/wiki/9.7inch_e-Paper_HAT, the chinese version has a bit more info and code: http://www.waveshare.net/wiki/9.7inch_e-Paper_HAT

joukos, 2018-11-03:
About the LUTs - I *guess* they're in the controller with no way to change them, as the [datasheet](http://www.waveshare.net/w/upload/1/18/IT8951_D_V0.2.4.3_20170728.pdf) mentions 8 programmable LUTs, but the command list seems very sparse and no mention if these can be reprogrammed (or read) easily from user code.

There's a small baby tugging at my trouser leg so I don't have time to look at this further right now, but seems like an interesting display for sure if it works as advertised.

phaer, 2018-11-03:
> Perhaps  you can ask Waveshare themselves about it, I've gotten quick replies  from their support earlier, and even some sample code that I haven't  seen elsewhere - the Wiki might not have it all yet.

I did, they confirmed that it supports partial refresh, but didn't provide example code so far - but I sent the last mail just yesterday, so we'll see.

> About the LUTs - I guess they're in the controller with no way to change them, as the datasheet mentions 8 programmable LUTs, but the command list seems very sparse and no mention if these can be reprogrammed (or read) easily from user code.

Okay, thanks for the confirmation, I wasn't sure if there's really nothing about reading them in the documentation or if I am missing something - I don't have much experience with low level hardware programming. And I found it strange, that the examples all used mode 2 and the docs did not seem to mention what they other modes were. But _right_ after posting this issue, I found a document which describes the mode, a bit hidden in the chinese wiki (thanks google translate!) http://www.waveshare.net/w/upload/c/c4/E-paper-mode-declaration.pdf

phaer, 2018-11-05:
I've got partial refresh working :) Going to look into PaperTTY support for this mode, but it might take a while as I have limited time available for this project.

joukos, 2018-11-08:
Neat! :)

JBShooster, 2019-02-20:
I have the Waveshare 6in model. It is quite similar to the 9.7in as it also runs on the IT8951 driver. Right now, trying to figure out how to get PaperTTY working on this device. Also, trying to determine how you accessed the driver functions in Python.

joukos, 2019-02-25:
> I have the Waveshare 6in model. It is quite similar to the 9.7in as it also runs on the IT8951 driver. Right now, trying to figure out how to get PaperTTY working on this device. I also want to figure out partial-refresh.

Nice, do mention if you get it working. At some point it would be nice to add support for these since they seem to be superior to the older ones.

phaer, 2019-02-25:
I stopped using PaperTTY as a base and tried to get a local https://www.freedesktop.org/wiki/Software/kmscon/wlterm/ fork to output on my display because it's a lot faster and a fully-featured terminal emulator. Then I paused the project for personal reasons. Going to try to publish some example code during the week, sorry for the delay

joukos, 2019-02-27:
Nice, I hope you get it to work, it would be really cool for these new displays to be put to good use.

I'm still a bit inclined to try implementing a "PaperVNC" viewer or similar, so that "everything" would work with it - if I ever get the time to continue with these experiments.

skipper113, 2019-03-29:
Cool, got myself a waveshare 9.7' and am interested in it as well. Looking forward for some example codes!

dvn0, 2019-04-30:
@phaer did you get anywhere with wlterm?

ekerebel, 2019-05-18:
@phaer, @joukos: I have been playing with the 6inch EPD and the IT8951 Hat for my project, successfully managing partial refreshes. My next step is to create my own PCB, using the IT8951. The question I have: do I need to program firmware/ load the LUT or do these come with the controller?

mchack23, 2019-06-04:
@ekerebel @phaer: Hi I've also gotten the 6inch EPD and the IT8951 Hat after reading this thread and I "simply" want a terminal running on it ;) but in fact I have no Idea how to run papertty on it. Which driver did you use? What did you do to have partial refreshes on it? All I can do for now is compile the IT8951.c example from waveshare and show a .bmp on the display but don't know a way of getting papertty running on it. Any pointers?

phaer, 2019-06-04:
Sorry for letting you all hanging for so long. Life came in between and I shelved the project for the last month, got my code base in a backup and going to try to look into it to answer your questions in the next weeks, but it could still take a while

Mozartenhimer, 2019-06-08:
> I've got partial refresh working :) Going to look into PaperTTY support for this mode, but it might take a while as I have limited time available for this project.
I bought the six inch model, and it looks like the 9.7" and 6" both have the new IT8951 controller.  

I'm looking forward to seeing that 9.7" example code!
 In the mean time, I'm going to try to get it working with the 6".
I have this dream of not having to stare at a glowing display to code, and I want to make that a reality.

Off topic, but has has anyone found the IT8951 firmware source code?

GregDMeyer, 2019-08-26:
Hi all,

I implemented a Python driver for the IT8951 controller: https://github.com/GregDMeyer/IT8951

I also have been working on a fork of PaperTTY using that driver, but I haven't uploaded it yet---have to do a bit of cleanup first. Hopefully in the next few days! 

Hopefully people find it useful :)

tg-x, 2019-08-26:
Just checked, they also have now higher resolution (1872×1404) & faster (full refresh 450ms) [7.8"](https://www.waveshare.com/product/modules/oleds-lcds/e-paper/7.8inch-e-Paper.htm) & [10.3"](https://www.waveshare.com/product/modules/oleds-lcds/e-paper/10.3inch-e-Paper-D.htm) displays -- anyone tried these yet?

Something like a [PineTab](https://www.pine64.org/pinetab/) with an e-ink display like these would be great..

mchack23, 2019-09-04:
> I implemented a Python driver for the IT8951 controller: https://github.com/GregDMeyer/IT8951
> 
> I also have been working on a fork of PaperTTY using that driver, but I haven't uploaded it yet---

yay, thank you very much! I've been able to run your test.py successfully printing on my 6" waveshare display. But since the thing I really want is having a terminal on that display I'm eagerly awaiting your fork (or a hint on how to use your driver with the master branch of papertty.
just to let you know your work (and of course joukos' work) is highly appreciated and eagerly awaited :)

GregDMeyer, 2019-09-05:
Thanks for the bump @mchack23 ! 

Here is my fork of PaperTTY for the IT8951: https://github.com/GregDMeyer/PaperTTY
It is quite heavily modified in order to take advantage of a lot of the features of the IT8951 controlled displays. Hopefully people enjoy it; would love any feedback! I think it probably doesn't make sense to try to merge that repo with this one (they are sufficiently different). But I did implement a couple things that perhaps could get pulled into this repo---for example parsing character attributes from vcsa (i.e. bold, background/foreground colors).

That being said, if folks want to use the original PaperTTY instead---it would be pretty trivial to add the IT8951 driver to the set of drivers included here, as well. I can do that and submit a pull request if people are interested. :)

joukos, 2019-09-05:
@GregDMeyer nice! If you get around to making the PRs, I'll gladly merge them.

phaer, 2019-09-05:
@GregDMeyer Wow, thanks a lot for sharing this! Running it right now and its pretty usable :)

mchack23, 2019-09-05:
@GregDMeyer this is working beautifully. Thanks a lot! 

I had two minor issues that were solved more or less quickly (just listing them here if someone else encounters those): 
1. I hadn't set the correct resolution in raspi-config and it took me a while to realize that there wasn't any option in papertty to set the right rows and colums (had not enough rows (quarter bottom display was blank) but text spilled over in the colums) but that in fact it was all because the resolution wasn't set to 800x600 in raspi-config. after that it worked flawlessly.
2. can't scroll up using shift-pgup as I'm used to ... don't know the reason yet but as a workaround I just use tmux and then ctrl-b pgup to scroll up :)
---
Else everything works as I wished and I really am quite happy with my little eink terminal. It is a bit slow still, so I wonder if there is an option possible that would forget about greyscale etc and just try and be as fast as possible. (like the --flip argument, a --fast for fast mode ;) ... but well that's nitpicking already. It is pretty usable just as phaer commented and I'm already dreaming of getting the huge screen from waveshare and putting it in an old laptop case with a raspi :) thank you guys so much! I've been dreaming of having this since I had my first kindle (it was a 3g and had a keyboard and I had a terminal on it, but I wanted a real keyboard and a real linux terminal with ssh and tmux and everything and now I have just that :D

GregDMeyer, 2019-09-05:
so awesome to hear! I am wary to discuss too much on this thread here, but if you want to open an issue on my repo I will comment there!! :) real quick response regarding the speed---there are still some optimizations to be had in communicating with the device, that I hope to implement soon. Hopefully then it will be a bit snappier!

fimad, 2019-10-21:
Is anyone actively working on porting @GregDMeyer's changes to this version of the library? If not, I have some local changes that add IT8951 support to this version of PaperTTY that I can upload.

GregDMeyer, 2019-10-29:
@fimad sorry to not push those changes yet. I'm happy to write a quick function that calls my driver, though if you have a standalone implementation for this library that might be better so that people don't have to install a second package to get it working.

fimad, 2019-11-01:
Sounds good, I have a standalone version that doesn't add any new dependencies. I'll clean it up and send out a PR this weekend.

joukos, 2019-11-03:
Thanks to @fimad the support for IT8951 is now merged (https://github.com/joukos/PaperTTY/commit/7e405bfd7e4748da39328509f2ee0f2d16fa3a11) - please test it with the 9.7" :)

P.S. I welcome any screenshots with these displays that could be added to the main README. Also, I haven't found a single video online demonstrating a Linux CLI/GUI (or pretty much anything useful) with any of these displays, so I've hesitated getting one before seeing how the update speed is etc. It would be cool if someone could make a short demonstration clip :)

joukos, 2019-11-03:
It appears the 9.7" works with the latest code, great work and thanks to you all. I'll close this issue now as it's pretty long already, so if there's any problems new issues should be created.

closed 2019-11-03 by joukos

sergarcia, 2020-09-07:
> It appears the 9.7" works with the latest code, great work and thanks to you all. I'll close this issue now as it's pretty long already, so if there's any problems new issues should be created.

I read through the post and it is unclear to me if the partial refresh that some of the commenters got working was actually implemented. In the current version of paperTTY, does the 9.7 inch display refresh quick enough to be usable as an interactive terminal? (According to the manufacturer the partial refresh rate is about .3s, which appears that would be sufficient for terminal use)

joukos, 2020-09-12:
I *think* (all?) IT8951 driven displays support partial refreshes, but the current method of handling the SPI writes with Python is not optimal and as the larger displays typically have bigger resolutions too, in addition to the marketed 0.3s it may take several seconds to transfer the data for the update. The only IT8951 display I have is the 6" HD and sometimes it seems to update pretty fast and sometimes not, so YMMV. Whether it's usable depends on your definition of interactive, but I'd say that with the current implementation the bigger displays are better suited for things that don't update too often. That said, if you've used the internet with dial-up and didn't complain, it might be bearable, but I wouldn't expect a very responsive result with the 9.7". I haven't tried one though. It also depends which RPi you're using, the beefier the better, but in the end frame data transfer is the bottleneck.

Regarding SPI, there was some work done here https://github.com/joukos/PaperTTY/issues/36#issuecomment-629889299 which would be nice to incorporate into PaperTTY, but time hasn't allowed to look at that yet in detail. It sounds like that would help a lot with the refresh speeds.