joukos/PaperTTY #28: Tested on Waveshare 2.7 inch epaper Pi Hat but issue with external Keyboard Input

mauricecyril, 2019-02-08:
Just tested on the Waveshare 2.7 Inch epaper Pi Hat (https://www.waveshare.com/wiki/2.7inch_e-Paper_HAT) and can confirm everything installed and that the terminal commands work when using ssh. The script to run at boot also works however I'm not able to use an external keyboard (screen remains static in the login prompt and all key presses are not reflected or refreshing). Was there anything that you e configured to allow usb keyboard inputs to show up on the mirrored terminal?

mauricecyril, 2019-02-09:
Opps looks like I didn't read your instructions all the way till the end. Keyboard input works now if I press Ctrl + Alt + F1 on the keyboard.

mauricecyril, 2019-02-09:
After some further testing I can confirm that the 2.9inch Waveshare does not support partial refresh.

joukos, 2019-02-25:
Thanks for testing.

closed 2019-02-25 by joukos