joukos/PaperTTY #33: Partial Refresh for waveshare v2?

neutralinsomniac, 2019-11-05:
I have very limited experience with all of this, but the pwnagotchi project has partial refresh working great on the waveshare v2 and I was thinking maybe the code could be adapted here?

The relevant code:
https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/ui/hw/libs/waveshare/v2/waveshare.py

joukos, 2019-11-05:
Yeah, it ought to be possible and I think the current code just lacks a few details for it to work like the demo code. Unfortunately I don't personally have this V2 variant (nor much time to spend on this) so I can't really test it. Contributions are welcome as always, I think it should be a fairly simple fix.

(Apparently I had taken a quick look when the basic support was merged https://github.com/joukos/PaperTTY/pull/30#issuecomment-514096557)

plan5, 2020-09-25:
I've closed my initial pull request after noticing some unexpected behavior (almost working, though).

Currently, partial refresh creates a distorted image on the first run. 
Running again after a clean exit, terminal produces clear text.

I've re-forked and commited my complete changes here: 
https://github.com/plan5/PaperTTY/commit/78243beaffeb0b49ff97b3d35a0f7b6dfbb62d43

I don't know how to go on from here. Should I create a pull request anyway as this is a partial progress?

joukos, 2020-10-04:
@plan5 if it's still very experimental, maybe making a PR can wait until the any showstoppers are fixed so that it wouldn't stay too long in a pending state (such as the other simple PR I simply haven't had time to finish...). Your description of the problem sounds like in a lucky case it _might_ be a fairly simple fix, though, like adjusting how the display is initialized, but unfortunately it's something I can't help much with right now.