joukos/PaperTTY #34: ERROR: Failed building wheel for Pillow

chinmoyrick, 2019-11-22:
ERROR: Failed building wheel for Pillow

plz help me to solve this....

`pi@raspberrypi:~/PaperTTY $ mkvirtualenv -p /usr/bin/python3 -r requirements.txt papertty
Already using interpreter /usr/bin/python3
Using base prefix '/usr'
New python executable in /home/pi/.virtualenvs/papertty/bin/python3
Also creating executable in /home/pi/.virtualenvs/papertty/bin/python
Installing setuptools, pkg_resources, pip, wheel...done.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting Pillow
  Downloading https://files.pythonhosted.org/packages/5b/bb/cdc8086db1f15d0664dd22a62c69613cdc00f1dd430b5b19df1bea83f2a3/Pillow-6.2.1.tar.gz (37.7MB)
     |████████████████████████████████| 37.7MB 3.5kB/s
Collecting RPi.GPIO
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.0-cp35-cp35m-linux_armv6l.whl (67kB)
     |████████████████████████████████| 71kB 162kB/s
Collecting spidev
  Downloading https://www.piwheels.org/simple/spidev/spidev-3.4-cp35-cp35m-linux_armv6l.whl
Collecting click
  Downloading https://www.piwheels.org/simple/click/Click-7.0-py2.py3-none-any.whl (81kB)
     |████████████████████████████████| 81kB 128kB/s
Collecting vncdotool
  Downloading https://www.piwheels.org/simple/vncdotool/vncdotool-0.13.0-py3-none-any.whl
Collecting Twisted
  Downloading https://www.piwheels.org/simple/twisted/Twisted-19.10.0-cp35-cp35m-linux_armv6l.whl (3.1MB)
     |████████████████████████████████| 3.1MB 72kB/s
Collecting Automat>=0.3.0
  Downloading https://files.pythonhosted.org/packages/e5/11/756922e977bb296a79ccf38e8d45cafee446733157d59bcd751d3aee57f5/Automat-0.8.0-py2.py3-none-any.whl
Collecting constantly>=15.1
  Downloading https://files.pythonhosted.org/packages/b9/65/48c1909d0c0aeae6c10213340ce682db01b48ea900a7d9fce7a7910ff318/constantly-15.1.0-py2.py3-none-any.whl
Collecting hyperlink>=17.1.1
  Downloading https://files.pythonhosted.org/packages/7f/91/e916ca10a2de1cb7101a9b24da546fb90ee14629e23160086cf3361c4fb8/hyperlink-19.0.0-py2.py3-none-any.whl
Collecting zope.interface>=4.4.2
  Downloading https://www.piwheels.org/simple/zope-interface/zope.interface-4.7.1-cp35-cp35m-linux_armv6l.whl (163kB)
     |████████████████████████████████| 163kB 7.1kB/s
Collecting PyHamcrest>=1.9.0
  Downloading https://www.piwheels.org/simple/pyhamcrest/PyHamcrest-1.9.0-py2.py3-none-any.whl (52kB)
     |████████████████████████████████| 61kB 32kB/s
Collecting attrs>=17.4.0
  Downloading https://www.piwheels.org/simple/attrs/attrs-19.3.0-py2.py3-none-any.whl
Collecting incremental>=16.10.1
  Downloading https://files.pythonhosted.org/packages/f5/1d/c98a587dc06e107115cf4a58b49de20b19222c83d75335a192052af4c4b7/incremental-17.5.0-py2.py3-none-any.whl
Collecting six
  Downloading https://www.piwheels.org/simple/six/six-1.13.0-py2.py3-none-any.whl
Collecting idna>=2.5
  Downloading https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl (58kB)
     |████████████████████████████████| 61kB 235kB/s
Requirement already satisfied: setuptools in /home/pi/.virtualenvs/papertty/lib/python3.5/site-packages (from zope.interface>=4.4.2->Twisted->vncdotool->-r requirements.txt (line 5)) (41.6.0)
Building wheels for collected packages: Pillow
  Building wheel for Pillow (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.virtualenvs/papertty/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-cz_wrs2v/Pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-cz_wrs2v/Pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-c7l05ek4 --python-tag cp35
       cwd: /tmp/pip-install-cz_wrs2v/Pillow/
  Complete output (172 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv6l-3.5
  creating build/lib.linux-armv6l-3.5/PIL
  copying src/PIL/ImImagePlugin.py -> build/lib.linux-armv6l-3.5/PIL
  copying src/PIL/ImageQt.py -> build/lib.linux-armv6l-3.5/PIL
  copying src/PIL/TiffImagePlugin.py -> build/lib.linux-armv6l-3.5/PIL
  copying src/PIL/ImageDraw.py -> build/lib.linux-armv6l-3.5/PIL
  copying src/PIL/PcdImagePlugin.py -> build/lib.linux-armv6l-3.5/PIL
  copying src/PIL/ContainerIO.py -> build/lib.linux-armv6l-3.5/PIL
  copying src/PIL/IptcImagePlugin.py -> build/lib.linux-armv6l-3.5/PIL
  copying src/PIL/_tkinter_finder.py -> build/lib.linux-armv6l-3.5/PIL
  copying src/PIL/ImageOps.py -> build/lib.linux-armv6l-3.5/PIL
[... 317 lines cut ...]
      File "/usr/lib/python3.5/distutils/command/build_ext.py", line 339, in run
        self.build_extensions()
      File "/tmp/pip-install-cz_wrs2v/Pillow/setup.py", line 687, in build_extensions
        raise RequiredDependencyException(f)
    __main__.RequiredDependencyException: jpeg

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-cz_wrs2v/Pillow/setup.py", line 903, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:

    The headers or library files could not be found for jpeg,
    a required dependency when compiling Pillow from source.

    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html

    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.virtualenvs/papertty/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-cz_wrs2v/Pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-cz_wrs2v/Pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-161nk5do/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.virtualenvs/papertty/include/site/python3.5/Pillow Check the logs for full command output.
(papertty) pi@raspberrypi:~/PaperTTY $ ^C
(papertty) pi@raspberrypi:~/PaperTTY $ dir
`

chinmoyrick, 2019-11-22:
its look like that..
`(papertty) pi@raspberrypi:~/PaperTTY $ sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py list
Traceback (most recent call last):
  File "./papertty.py", line 14, in <module>
    import drivers.drivers_base as drivers_base
  File "/home/pi/PaperTTY/drivers/drivers_base.py", line 18, in <module>
    from PIL import Image
ImportError: No module named 'PIL'
`

joukos, 2019-11-22:
Hmm... did you install the `libopenjp2-7` package? Is it running Raspbian or similar?

chinmoyrick, 2019-11-23:
pip install Pillow==2.2.2 
i think it can solve that error . 
can i replace it in requerement.txt ?

chinmoyrick, 2019-11-23:
hi.
i want to display terminal . i am using 2.13 c epaper display . 
i replace Pillow==2.2.2 in requirements.txt . after that it install sucsessfully.
`Successfully installed Automat-0.8.0 Pillow-2.2.2 PyHamcrest-1.9.0 RPi.GPIO-0.7.0 Twisted-19.10.0 attrs-19.3.0 click-7.0 constantly-15.1.0 hyperlink-19.0.0 idna-2.8 incremental-17.5.0 six-1.13.0 spidev-3.4 vncdotool-0.13.0 zope.interface-4.7.1
`
next i try `sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py list`
it also working and i try with EPD2in13b driver and got some error
`(papertty) pi@raspberrypi:~/PaperTTY $ sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py --driver EPD2in13b scrub
Traceback (most recent call last):
  File "./papertty.py", line 513, in <module>
    cli()
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/decorators.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "./papertty.py", line 365, in scrub
    ptty.driver.scrub(fillsize=size)
  File "/home/pi/PaperTTY/drivers/drivers_base.py", line 60, in scrub
    self.fill(self.black, fillsize=fillsize)
  File "/home/pi/PaperTTY/drivers/drivers_base.py", line 67, in fill
    self.draw(x, 0, image)
  File "/home/pi/PaperTTY/drivers/drivers_color.py", line 41, in draw
    self.display_frame(self.get_frame_buffer(image))
  File "/home/pi/PaperTTY/drivers/drivers_colordraw.py", line 475, in get_frame_buffer
    super().get_frame_buffer(image, reverse=reverse)
  File "/home/pi/PaperTTY/drivers/drivers_full.py", line 76, in get_frame_buffer
    ({0}x{1}).'.format(self.width, self.height))
ValueError: Image must be same dimensions as display                 (104x212).`

plz help me what to do next .
**i want to display one webpage at raspberry pi start up**

joukos, 2019-11-23:
Does it work with either of these commands, without the `--scrub`:

- `sudo ./papertty.py --driver EPD2in13b terminal`
- `sudo ./papertty.py --driver EPD2in13b terminal --portrait`

If you intend to display a web page, I suppose you'll want VNC anyway, so you might try that instead. The terminal orientation code is sloppy and unfinished and the VNC part of the code is more straightforward in that area, so it might work even if the terminal doesn't.

For reference, a similar issue: https://github.com/joukos/PaperTTY/issues/19

chinmoyrick, 2019-11-24:
First i try with vnc 
pi@raspberrypi:~/PaperTTY $sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py --driver epd2in13b vnc --display 1 --password chinmoy --sleep 0.1 --rotate 90
initial (1): (104, 212)
^C
Aborted!

**it flash display but can't display anything meaningful.**

**try to do same thing with a fresh installed raspberrypi** but got same error . 
fallowing your installation process 

(papertty) pi@raspberrypi:~/PaperTTY $ sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py --driver EPD2in13b terminal --portrait
Started displaying /dev/vcsa1, minimum update interval 0.1 s, exit with Ctrl-C
Traceback (most recent call last):
  File "./papertty.py", line 513, in <module>
    cli()
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/decorators.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "./papertty.py", line 498, in terminal
    **textargs)
  File "./papertty.py", line 255, in showtext
    draw.text((0, 0), text, font=self.font, fill=fill, spacing=spacing)
TypeError: text() got an unexpected keyword argument 'spacing'
`Requirements
Enable SPI (sudo raspi-config)
Interfacing Options -> SPI -> Yes
Reboot
Steps
Clone the repo somewhere and enter the directory
git clone https://github.com/joukos/PaperTTY.git
cd PaperTTY
Install virtualenv and libopenjp2
sudo apt install virtualenvwrapper python3-virtualenv libopenjp2-7
Source the wrapper to use mkvirtualenv (you may want to add this to ~/.bashrc)
source /usr/share/virtualenvwrapper/virtualenvwrapper.sh
Create the Python 3 virtualenv and install packages in requirements.txt
mkvirtualenv -p /usr/bin/python3 -r requirements.txt papertty
This will create ~/.virtualenvs/papertty which contains the required environment
After creating the virtualenv, it should become active and you should see (papertty) on your prompt
Note: the software needs to be run with sudo in the typical case, so you need to explicitly start the interpreter within the virtualenv - otherwise the program attempts to import system packages instead
You should now be able to run sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py list to see the`
**i just fallow those steps**
Sorry i don't know .. and thanks for your reply ..

joukos, 2019-11-24:
The `spacing` problem I experienced with some installation once, ie. the library didn't have that - is this Raspbian you're working with?

The VNC command sounded promising since it didn't crash for you, so maybe try a couple of more things with it:

- First of all, this is (as far as I know) a **slow** display that doesn't support partial refresh - you probably don't want to use `--sleep 0.1` with it, which would mean that the code would only wait 100ms before starting a full refresh cycle
- Ensure your VNC desktop actually has some program running that will show up when drawn
- Try something like: `sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py --driver epd2in13b vnc --display 1 --password chinmoy --sleep 30 --rotate 90`

The terminal command you might be able to fix by editing this line: https://github.com/joukos/PaperTTY/blob/master/papertty.py#L255 - change it to `draw.text((0, 0), text, font=self.font, fill=fill)`.

chinmoyrick, 2019-11-24:
nop after edit 255 line  edit
`pi@raspberrypi:~/PaperTTY $ sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.                                                                                        py --driver epd2in13b vnc --display 1 --password chinmoy --sleep 30 --rotate 90
Traceback (most recent call last):
  File "./papertty.py", line 513, in <module>
    cli()
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py                                                                                        ", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py                                                                                        ", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py                                                                                        ", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py                                                                                        ", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/core.py                                                                                        ", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/click/decorat                                                                                        ors.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "./papertty.py", line 404, in vnc
    ptty.showvnc(host, display, password, int(rotate) if rotate else None, inver                                                                                        t, sleep, fullevery)
  File "./papertty.py", line 206, in showvnc
    client.refreshScreen()
  File "/home/pi/.virtualenvs/papertty/lib/python3.5/site-packages/vncdotool/api                                                                                        .py", line 108, in proxy_call
    raise VNCDoException(result)
vncdotool.api.VNCDoException: [Failure instance: Traceback (failure with no fram                                                                                        es): <class 'twisted.internet.error.ConnectionRefusedError'>: Connection was ref                                                                                        used by other side: 111: Connection refused.
]
`
just use as normal ---       pi@raspberrypi:
i dont know how to get this (papertty) pi@raspberrypi:

chinmoyrick, 2019-11-24:
i am using 2.13 inch (c) W module 
and fallowing process..
Enable SPI ,
git clone https://github.com/joukos/PaperTTY.git
cd PaperTTY
edit Pillow==2.2.2 in requerement.txt 
sudo apt install virtualenvwrapper python3-virtualenv libopenjp2-7
source /usr/share/virtualenvwrapper/virtualenvwrapper.sh
mkvirtualenv -p /usr/bin/python3 -r requirements.txt papertty
sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py list
(papertty) pi@raspberrypi:~/PaperTTY $ sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py --driver EPD2in13b terminal --portrait
thats all ..

joukos, 2019-11-24:
`Connection refused.` means your VNC server isn't running or it can't be connected to - ensure it's running and try again.

Also, removing the spacing argument only affects the `terminal` feature, not VNC.

chinmoyrick, 2019-11-24:
`pi@raspberrypi:~/PaperTTY $ vncserver -geometry 250x128 :1

New 'X' desktop is raspberrypi:1

Starting applications specified in /home/pi/.vnc/xstartup
Log file is /home/pi/.vnc/raspberrypi:1.log

pi@raspberrypi:~/PaperTTY $ sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py --driver epd2in13b vnc --display 1 --password chinmoy --sleep 30 --rotate 90
initial (1): (104, 212)
full (2): (104, 212)
[image]
`

joukos, 2019-11-24:
Since it doesn't crash or complain now and reports having done refreshes, it seems to be working. Have you started anything on the VNC display? That is, maybe running something like `DISPLAY=:1 xeyes` in another shell to see if it appears on the screen?

As a sidenote, you're starting a 250x128 VNC desktop but your display is 212x104 - this is fine since the program rescales it to fit the screen geometry, but your graphics will appear squished and you could change the VNC geometry to match the display to avoid this.

chinmoyrick, 2019-11-24:
sorry i don't know what to do . and thank you for your grate support . i like your project so much. grate job . its not working for me.. whatever thank you so much for your valuable time . have a good day ,..
take care ...

closed 2019-11-24 by chinmoyrick