joukos/PaperTTY #43: Clean up and organize README

joukos, 2020-01-12:
The main README is too long. There's old but valid documentation way too far to be comfortably scrolled to because of meandering hit-and-run updates and not having the time to keep the docs concise and updated.

Some stuff to do:
- VNC feature should be included in the command line instructions
   - VNC instructions could be a bit more thorough, too
- Images are nice to have, but they could be clickable thumbnails in tables (or something) so they won't take most of the space
- Clean up, fix historical remarks

This would be easiest for me to do, but since it's so hard to get a proper moment to do it, I've created this issue to nag me. And if I never get around to it, maybe someone else with a knack for documentation will :)

chi-lambda, 2020-01-13:
Github has a wiki feature. Would that be a better format?

joukos, 2020-01-13:
In some other old project I used the Wiki, but the problem is that no one looks there :) If there's a whole library of information, it might be useful, but I think in this case it would be more effective to clean up the main README.