joukos/PaperTTY #44: Create PyPi package

joukos, 2020-01-12:
It would be very nice if it was possible to simplify the installation of PaperTTY to just `pip install papertty`. Any help with achieving this is appreciated.

colin-nolan, 2020-05-09:
I'd be happy to take a look at this!

Unless you have an objection, I would setup the project with a `pyproject.toml` (as per PEP 518), which can be nicely managed with a tool such as [poetry](https://python-poetry.org/).

I suspect it will be easiest for `papertty.py` to be moved out of the root and into a new `papertty` package.

joukos, 2020-05-14:
You're more than welcome to :) I almost started with this some time ago but didn't have time to do anything concrete.

PR #61 (2020-08-23): Packaging and PyPI distribution
This PR sets up packaging with a [`pyproject.toml` file](https://www.python.org/dev/peps/pep-0518/) and the help of [Poetry](https://python-poetry.org/).

I have tried to change as little as possible to get this to work. However, it was necessary to create a `papertty` package (directory) and move `papertty.py` + the drivers into that because I don't think it's possible to have a single module in addition to a package (which in this case was the `drivers` package). 

I've setup [`papertty` on PyPI](https://pypi.org/project/papertty/), which I will pass control o [... cut]
closed 2020-08-23 by joukos