joukos/PaperTTY #48: Safety and image clarity

joukos, 2020-01-27:
In order to prevent any potential damage for long running applications and keep the image clean, a couple of things ought to be done:

- [ ] Set the panel to deep sleep after a refresh
- [ ] Apply a full-refresh update every now and then for `terminal` mode (`vnc` does this already)
   - [ ] Also allow to manually initiate (such as `scrub` currently - but it's not as useful)

Rationale for deep sleep is an entry in the Waveshare wiki (not sure when it was added but I don't remember seeing it earlier):

> Question:
>Why my e-paper has ghosting problem after working for some days
>Answer:
>Please set the e-paper to sleep mode or disconnect it if you needn't  refresh the e-paper but need to power on your development board or  Raspberry Pi for long time.Otherwise, the voltage of panel keeps high  and it will damage the panel

gdkrmr, 2020-01-27:
Handle `SIGUSR1` for full refresh and `SIGUSR2` for a short deep sleep?

chi-lambda, 2020-01-27:
My 6" display doesn't seem to have that problem. Might be a common feature of IT8951 displays, thanks to the additional controller.

@math85360 @jeLee6gi @C-Rothnie, what is your experience?

joukos, 2020-01-27:
> Handle `SIGUSR1` for full refresh and `SIGUSR2` for a short deep sleep?

I think I would put the display to sleep automatically after N updates with no image changes, if only it's feasible to do so (ie. no unreasonable overhead or other issues). N could even be quite low, or a particular elapsed time. The signals can be left for manual overrides, but I'm not yet decided on what those should be.

I'm thinking that perhaps there should be a `--powersave` option (on by default) that does at least two things:

1. Puts the display to sleep when nothing is happening
1. Automatically adjusts the sleep interval to something larger (up to a certain maximum for example)

This would keep the display mostly asleep if the updates happen irregularly and also keep the CPU usage down in those cases by making PaperTTY itself slow its pace. When the image updates again, it would go back to regular refresh rates (or perhaps there could be a slowly adaptive method for this).

To do this nicely we might need to have some configurable values such as the minimum and maximum update intervals.

So, maybe in the future we could have `--powersave` with perhaps settings like:
- `off` - no powersaving, static and minimum update interval (basically the way it works currently).
- `on` - enable panel sleep after a while (N frames or reasonable wall clock time of nothing happening?) and increase the interval a reasonable amount (say, from 0.1 seconds to at least 1 second). When something happens, panel is woken up and the interval is immediately set back to minimum.
- `adaptive`- enable panel sleeping and adjust update interval automatically, this could start from the fastest update we want to have and then gradually adjust it slower depending how often actual updates occur (ie. change in the image). For example a calendar application doesn't need to update but once an hour perhaps in some cases, but since the loop isn't too costly maybe the max interval can be set to some minutes. If suddenly there's a flurry of activity, the interval could be shortened gradually. The algorithm to do this should probably be kept pretty simple to understand.
- something else?

I've been meaning to tackle the config file issue at some point, perhaps as "profiles" in YAML format (so one can do `./papertty.py -f cottage_vnc.yaml` etc.), and with all these new configuration options it's going to be needed soon...

Dunno, how does this sound?

lvan188, 2020-10-25:
Can you post a tutorial video

joukos, 2020-10-25:
> Can you post a tutorial video

Please create a new issue about this if you feel it's needed (requesting it in an unrelated issue is confusing) - the documentation has mostly been rewritten already, but as usual, I've been too busy with other things to finish it.