joukos/PaperTTY #50: trouble running papertty.py

skywo1f, 2020-04-09:
File "./papertty.py", line 716, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "./papertty.py", line 502, in vnc
    ptty = settings.get_init_tty()
  File "./papertty.py", line 394, in get_init_tty
    tty.init_display()
  File "./papertty.py", line 246, in init_display
    self.driver.init(partial=self.partial)
  File "/home/pi/PaperTTY/drivers/drivers_partial.py", line 69, in init
    if self.epd_init() != 0:
  File "/home/pi/PaperTTY/drivers/drivers_base.py", line 184, in epd_init
    self.SPI = spidev.SpiDev(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

skywo1f, 2020-04-09:
turns out I had forgotten to enable spi. I am now however getting this issue:
vncdotool.api.VNCDoException: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionRefusedError'>: Connection was refused by other side: 111: Connection refused.

skywo1f, 2020-04-09:
(papertty) pi@raspberrypi:~/PaperTTY $ sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py --driver epd2in13 vnc --display 1 --password supAPass --sleep 0.1 --rotate 90
Loading PIL font tom-thumb.pil. Font size is ignored.
Timeout to server localhost:1

joukos, 2020-04-10:
The VNC error messages could be made a bit more helpful. What command/settings are you using for the VNC server? Is the password correct?

ducloyer, 2020-04-24:
Hi
I'm trying to build a ebookreader with a pi0 and a 9.7inc waveshare epaper.
I probably do something wrong in the installation, but I don't find where.

I create de virtualenv with : 
mkvirtualenv -p /usr/bin/python3 -r requirements.txt papertty
No error appears in the windows

I create a connection with :
(papertty) pi@raspberrypi:~/Documents/github/PaperTTY-master $ vncserver -geometry 1200x825 :3
response :+1: 
Fichier journal : /home/pi/.vnc/raspberrypi:3.log
Nouveau bureau : raspberrypi:3 (192.168.1.22:3)
(papertty) pi@raspberrypi:~/Documents/github/PaperTTY-master $

But when I launched
sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py --driver it8951 vnc --display 3 --password supAPass --sleep 0.1 --rotate 90
The script end with 
Timeout to server localhost:3

It appened after the line 
client.refreshScreen()
in papertty.py

I already tried to forced the ip adress and the display, if i write a wrong adress or display I have another error, so I think it's ok

I also check the vncserver by connecting to it on my smartphone, it work's fine, but I noticed that there is a message on the desktop 
GDBus.Error:org.freedesktop.PolicyKit 1.Error.Failed: An authentication agent already exists for the given subject

I also tried to do the same without the virtualenv, it give the same error (timeout...), the only difference is that I have a warning at the begining 
:0: UserWarning: You do not have a working installation of the service_identity module: 'cannot import name 'verify_ip_address' from 'service_identity.pyopenssl' (/usr/lib/python3/dist-packages/service_identity/pyopenssl.py)'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.

ducloyer, 2020-04-24:
After 'vncserver -geometry 1200x825 :3' I launched 'x11vnc -display :3' and now I have my desktop on the epaper
I also tested without 'vncserver -geometry 1200x825 :3' and directly 'x11vnc -display :0', it works

joukos, 2020-04-25:
One thing to keep in mind is that the VNC server's password is separate from your user's (`pi`) or `root` password. Of course, you *can* set the same password but VNC typically might support only a maximum length of 8 and is not secure to begin with. Usually when starting up the server the first time it prompts for a password to set (and stores it in `~/.vnc/passwd` or so you don't need to set it again).

I'm not exactly sure what you achieved with running both `vncserver` and `x11vnc` but does it work the way you wanted now?

joukos, 2020-05-09:
I'll close this now - if you're still having trouble, please reopen the issue or create a new one.

closed 2020-05-09 by joukos