joukos/PaperTTY #57: ImportError: libtiff.so.5: cannot open shared object file: No such file or directory

rroels, 2020-05-09:
I followed the install instructions to the letter on a fresh install, but on the first run of I got the following error:
>(papertty) rroels@piz:~/PaperTTY $ sudo ~/.virtualenvs/papertty/bin/python3 ./papertty.py list
>Traceback (most recent call last):
>  File "./papertty.py", line 14, in <module>
>    import drivers.drivers_base as drivers_base
>  File "/home/rroels/PaperTTY/drivers/drivers_base.py", line 18, in <module>
>    from PIL import Image
>  File "/home/rroels/.virtualenvs/papertty/lib/python3.7/site-packages/PIL/Image.py", line 93, in > <module>
>    from . import _imaging as core
>**ImportError: libtiff.so.5: cannot open shared object file: No such file or directory**

Luckily the solution is trivial:
>sudo apt install libtiff5

Maybe this should be added to the install instructions?

joukos, 2020-05-10:
Hmm, I see, I guess in Raspbian the package is included by default. I'll add it, thanks!

joukos, 2020-05-10:
Fixed in https://github.com/joukos/PaperTTY/commit/2bc653d506cc576509990abe9d6478d714ab58fe.

closed 2020-05-10 by joukos