joukos/PaperTTY #66: Trouble getting papertty to run

sergarcia, 2020-09-10:
I have the 4.2 Waveshare display and I can run their python demo code fine. For papertty I have installed in a virtualenv as indicated in the readme. When I try the following command (from within the virtualenv)
`papertty --driver EPD4in2 image --image test-img.bmp`
, both as root or regular user, I get the following error:
~~~
Traceback (most recent call last):
  File "/home/pi/papertty_venv/bin/papertty", line 10, in <module>
    sys.exit(papertty.cli())
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/papertty/papertty.py", line 567, in image
    ptty = settings.get_init_tty()
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/papertty/papertty.py", line 398, in get_init_tty
    tty = PaperTTY(**self.args)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/papertty/papertty.py", line 75, in __init__
    self.driver = get_drivers()[driver]['class']()
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/papertty/drivers/drivers_4in2.py", line 72, in __init__
    super(drivers_partial.WavesharePartial, self).__init__(name='4.2"',
NameError: name 'drivers_partial' is not defined
~~~
Other attempts to run papertty produce similar errors.

joukos, 2020-09-12:
Ah, that's actually a little leftover from shuffling the code around - https://github.com/joukos/PaperTTY/blob/master/papertty/drivers/drivers_4in2.py#L72 should be `super(WavesharePartial, self).__init__(...` instead (wish we had some tests...). Thanks for pointing it out.

I pushed a fixed version (0.1.6) to PyPi just now. If you can test and verify it works for you now that'd be great.

sergarcia, 2020-09-13:
Awesome, that fixed it. I wonder how one would go about creating tests for a project like this involving hardware devices.

joukos, 2020-09-13:
Good to hear! I'll close this then.

Even without the hardware, most of the code could be tested automatically and bugs such as this would've been immediately found. One way to test the drivers too would be to check for example if the right sequences of bytes are written to SPI and whether the drawing commands produce proper results in a frame buffer, etc., but ultimately it of course needs someone to try it out with the actual hardware too.

closed 2020-09-13 by joukos