joukos/PaperTTY #68: Raspberry Pi 2 + 6inch HD e-Paper HAT

bofh, 2020-10-03:
Hello!

I’ve got Raspberry Pi 2 Model B V1.1 and IT8951 hat from Waveshare (https://www.waveshare.com/wiki/6inch_HD_e-Paper_HAT). Both this repository and version from @GregDMeyer have a trouble communicating to the HAT while this https://github.com/waveshare/IT8951-ePaper.git works.

Any suggestions how to debug this or configuration tips?

Thanks!

GregDMeyer, 2020-10-03:
there is a weird issue where in some cases if you run the waveshare code first, it causes this code to break. can you trying running the code in my repo right after rebooting the pi, and see if it works? good luck!

GregDMeyer, 2020-10-03:
(you can try running this one too, I just haven't looked at it in a while and I'm not sure how synced up we are any more)

joukos, 2020-10-04:
I have the 6" HD [too](https://github.com/joukos/PaperTTY/issues/32#issuecomment-586643096) and it has worked okay, though I haven't had time to play with it for some time. There were some initial problems where I had a power cable running on top of the flat flex and that seemed to disrupt the operation somehow... not 100% sure though if that actually was the case, but after some reorganizing it just worked.

> there is a weird issue where in some cases if you run the waveshare code first, it causes this code to break

If this is the case, maybe there's something that could be improved in the initialization of the display...

bofh, 2020-10-04:
Looks like this is the case, indeed. After clean reboot both versions work, after running Waveshare code initialization fails!

GregDMeyer, 2020-10-04:
glad its working! @joukos , yeah I'm not sure how it could be fixed. it seems that if you use the bcm2835 library to handle the SPI, the kernel no longer can control the SPI (my library talks to SPI thru the kernel now, not sure if this repo has been updated following that change a few months ago). I'm not sure if there's a way to give SPI back to the kernel without rebooting. if anyone knows the answer it would be great!