joukos/PaperTTY #69: Feature(?) Draw images when using the terminal

waynew, 2020-11-10:
Basic idea: With a regular TFT screen, you can be using the terminal, and then use `fbi` to view images on the TFT. If you quit, then it sends you back to the terminal.

I'm not sure if there's a way to do that with the existing codebase. I didn't find anything, and it appears that trying to write to the SPI/launch a second papertty process to draw an image just fails because the device is busy.

One imaginary approach is that the papertty process has some way of providing information, whether that's using inotify, or sending information via a port, or ???

I'm not sure exactly what process things currently use, but it would be really neat if that were a possibility.

joukos, 2020-11-11:
I've considered having some sort of control socket for interacting with the thing while it's running, which should make this pretty easy to do, but that hasn't been implemented yet. Might make sense to make it a bit more daemony overall to have more flexibility to do stuff like this.

a8ksh4, 2021-04-26:
One way to approach this is to stop the terminal service, show the image, and then start the terminal service again.

> cat view_img.sh
#!/bin/sh
sudo systemctl stop papertty
cat foo.png | sudo papertty --driver EPD4in2 image
read press_enter
sudo systemctl start papertty

But it would be way more cool to be able to see images in-line with text in the terminal...

hym3242, 2022-01-16:
This could be done with a framebuffer mode, in which papertty basically acts like a fbcp, read the fb0 and render just like how it does with vnc.

joukos, 2022-01-17:
@hym3242 actually added (a very preliminary version of) such a mode now https://github.com/joukos/PaperTTY/pull/86, been pending for a long time...