joukos/PaperTTY #70: waveshare 7.5inch HD supported?

assadollahi, 2020-12-11:
Thank you so for much for this AWESOME project that I have been following for a long time. I'm thinking about building a Psion-like portable computer and it would be great to use the 7.5in HD screen: https://www.waveshare.com/7.5inch-hd-e-paper-hat.htm can anybody tell me whether the software in this project supports it? If not, would the normal 800x600 resolution version work?

joukos, 2020-12-12:
Hmm, at least that doesn't use the IT8951 controller it seems (those have a good likelihood of working without writing a specific driver), but making it work should be doable similarly to the others by adapting the relevant parts of example code and creating a driver for it. Of the 7.5" models two are currently supported, V1 (640x384) and V2 (800x480). I think there's a 6" model with 800x600, but I don't know if anyone has one in order to write and test the driver. I have a 6" HD though, but that uses the IT8951 and is thus supported.

I haven't yet looked into the example code of this 7.5" HD and how similar it is to the existing ones, but it's currently not explicitly supported. If it's very similar to the other ones, supporting it might not be a lot of work. Writing drivers is not that much trouble usually, so if you already have the display and feel like doing that, contributions are always welcome. Generally speaking, if there's Python example code that can draw on the display, it can be made to work with PaperTTY.

Note however, that the 7.5" HD doesn't claim to support partial refresh and using it for anything very interactive is unlikely to be satisfying - even with partial refresh the bigger displays tend to be pretty slow because there's more data to transfer. You may want to reconsider if this is the model you want for the computer.

assadollahi, 2020-12-12:
Thanks so much for the details. So would you advise to go with the 7.5in V2 model (800x600), then? Or the 6in HD? My application is usually Geany or similar IDEs and/or a terminal.

joukos, 2020-12-13:
I guess it depends on how small text you want to read. Personally I think the 6" HD looks fantastic, but if you want to put every pixel to good use, text will look *super* tiny.

I actually just shot a little demo video because I wanted to try the display with a Raspberry Pi 400 to see how well it works in the software's current state: https://www.youtube.com/watch?v=OiK_xBshSjQ

The quality of the video is not so great since I shot it in dim lighting at an angle, but maybe it gives a bit of impression on how the system might work if it has a powerful enough Pi and such a display. Note that these IT8951 displays actually have multiple update modes and here the blinkiness is due to using grayscale - I think black and white should not flash and be faster also.

joukos, 2020-12-13:
Also, from those two your only feasible option is 6" HD as 7.5" V2 does not support partial refresh. Your best bet is probably an IT8951 display with partial refresh and pixel density you are comfortable with, but they also have the extra controller board and power requirements which may be a consideration. But partial refresh you absolutely need if you're going to mostly type.

assadollahi, 2020-12-27:
thanks a lot for your answers and expertise!

closed 2020-12-27 by assadollahi