joukos/PaperTTY #71: Partial refresh seems broken on rpi hat 2.13" v2

CrashdSystem, 2020-12-12:
I've just bought a pi Zero W and a 2.13" waveshare hat, looking forward to make a little distraction free word processor. I've managed to set almost all up, but oddly, using the driver for v2 I can only understand what is shown on the console when I launch a command with '--nopartial'. Otherwise, the screen only renders noise, or perhaps only some pixels become active.

I could still manage to use the device as intentded, as I want it to be sort of a reference of what I'm writing, and I don't think I'll be editing anything, but the full flashing is a bit annoying and the screen is supposed to be capable of partial refreshing.

Also, it is very likely that I am doin something wrong.

joukos, 2020-12-12:
Ah, yes, it seems it wasn't documented properly - the support for this one actually only covered full refresh and partial had some issues: https://github.com/joukos/PaperTTY/pull/30. Sorry about that, I think I've forgot to mention it anywhere (doc update is *still* pending...)

I'm not sure if the modification (maybe others needed too) I mentioned in the PR was ever tried and whether it might help, but might be worth a shot.

CrashdSystem, 2020-12-14:
Thanks for answering. I think I'll try https://github.com/plan5/PaperTTY/commit/78243beaffeb0b49ff97b3d35a0f7b6dfbb62d43 when I can

CrashdSystem, 2020-12-23:
Well, I just replaced the file that handles partial replies with https://github.com/plan5/PaperTTY/commit/78243beaffeb0b49ff97b3d35a0f7b6dfbb62d43 and seems to work quite well. Problem solved.

closed 2020-12-23 by CrashdSystem