joukos/PaperTTY #72: Trouble running VNC

thedogbeat, 2021-01-04:
I'm trying to make PaperTTY work through VNC, I have my Waveshare 9.7".
When running `papertty --driver it8951 vnc --display 1 --sleep 0.1 --rotate 90`
this is what happens:

`$ papertty --driver it8951 vnc --display 1 --sleep 0.1 --rotate 90
Loading PIL font /home/pi/.local/lib/python3.7/site-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
width = 1200
height = 825
img_addr = 0011a1e0
firmware = SWv_0.2.1T
lut = M841
VCOM = -2.00V
Timeout to server localhost:1
`
Sorry if I'm being lame (new to this)

joukos, 2021-01-04:
The VNC feature is a bit bare-bones at the moment with logging and error handling, so sometimes it might not be obvious what's wrong - I actually think the latest version might have a problem with a certain library not handling the password as it should, since this change got reverted at some point https://github.com/joukos/PaperTTY/commit/349edc2d4408a1f6a0f590b6f63da4e68cef6c6a but even though the vncdotool issue is closed, I think I encountered it again a while ago...need to check that.

Seems like in this case it just can't connect to the server though, but does the display at least work with `stdin` feature for example?

You should check the following:

- is your VNC server actually running? (ie. you have started it separately and it's running and listening on localhost - use `netstat -tlpn` to check it's there, and try to connect with another vncviewer to it to verify there's something visible)
- does the server have a password? If so, you should configure it on the command line with the `--password` option
- is the display number correct? `--display 1` means that we try to connect to a VNC server running on port `5901`, `--display 2` would be `5902` and so on

thedogbeat, 2021-01-06:
@joukos thanks for these details.
Unfortunately, even the `stdin` feature doesn't work for me either: the screen flashes but symbols don't print out (the waveshare wiki example works though). So I guess it's more basic stuff I face here.

I tried the installation steps and there's an error on the virtualenv creation:
`mkvirtualenv -p /usr/bin/python3 -i papertty requirements.txt papertty
usage: virtualenv [--version] [--with-traceback] [-v | -q] [--read-only-app-data] [--app-data APP_DATA] [--reset-app-data] [--upgrade-embed-wheels] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-posix,venv}]
                  [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list] [--clear] [--no-vcs-ignore] [--system-site-packages] [--symlinks | --copies] [--no-download | --download] [--extra-search-dir d [d ...]] [--pip version]
                  [--wheel version] [--setuptools version] [--no-pip] [--no-wheel] [--no-setuptools] [--no-periodic-update] [--symlink-app-data] [--prompt prompt] [-h]
                  dest
virtualenv: error: unrecognized arguments: papertty
SystemExit: 2
`
Also, 'sudo' doesn't work for me:
`sudo papertty --list
sudo: papertty: command not found`

So, I try to run it without 'sudo' from virtualenv (using the instructions from the readme beginning), or without it, but no difference.
Sorry for being lame, I guess I'm missing something simple.

joukos, 2021-01-10:
That looks like `mkvirtualenv` might be missing `-r` in front of `requirements.txt`, but this is actually a problem with the stale docs too - there's no `requirements.txt` either since moving to PyPi and poetry. What you might want to try instead is install `poetry` (https://python-poetry.org/docs/#installation) and then run `poetry install` in PaperTTY directory. This should make a virtualenv for you and install dependencies there, after which you should be able to `poetry shell` to get a shell with the virtualenv activated (see https://python-poetry.org/docs/basic-usage/#using-your-virtual-environment) and try to start `sudo papertty list` - this is a script that poetry should install in the virtual environment.

Let me know if that doesn't work at all, I've forgotten some poetry details and can't test it right now. And sorry for all the confusion, it's not you, it's the docs.

thedogbeat, 2021-01-11:
It ends up the same way with poetry.
When connecting through vncserver: `Timeout to server localhost:1`
When installing 'papertty' through `poetry` it was an error though: `Installing pillow (7.1.2): Failed`.
Do you think it can be related?

I also tried `stdin` and it worked! `Cowsay` displayed itself, so I guess it's a matter of connecting to VNC.
I did try to connect from the other machine to the same host `192.168.1.56:1` and it worked too!
Besided '--password' maybe I should also mention username too? Didn't find how to do it.

joukos, 2021-01-11:
Which VNC server software are you using? It could be something related to its authentication settings or other protocol details despite the "timeout" error. Since the VNC support is pretty barebones, my advice is to keep things as simple as possible, and I've usually used tightvncserver for testing it.

You have verified that the server listens on all addresses and not 192.168.1.56 only? (If you change localhost to 192.168.1.56, does it work then?)

Basic VNC authentication doesn't have a concept of usernames so just the password should be enough, but there's some fancier variants that do take a username and use system credentials etc. In case the server is configured to utilize something beyond a simple VNC password, it might not be supported by either vncdotool or the current way that PaperTTY utilizes it to connect to the server, which is minimalist at best.

Unless you're using it already, you might want to try `tightvncserver` instead:

```
sudo apt install tightvncserver
tightvncserver -localhost -geometry 1200x825  :1 # you'll be prompted to set a password
# now there should be a VNC display in localhost:1
# to kill the display, use `tightvncserver -kill :1`
```

Then try connecting to that. Edit `~/.vnc/xstartup` (created on first run) to start up some simple window manager or just a terminal or whatever to see that it works. I think by default the file is populated to start some session but I forget exactly.

joukos, 2021-01-31:
Closing this for now - feel free to reopen if the issue persists.

thedogbeat, 2021-02-01:
Sorry for bothering again, I see that everybody get this working, and I'm still having some strange issue.
Tried to re-install everything a few times, here's what's going on in case of using `tightvncserver`. The last line is the same though:

`papertty --driver it8951 vnc --display 1 --password ******** --sleep 0.1 --rotate 90
Loading PIL font /home/pi/papertty_venv/lib/python3.7/site-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
width = 1200
height = 825
img_addr = 0011a1e0
firmware = SWv_0.2.1T
lut = M841
VCOM = -2.00V
Unhandled Error
Traceback (most recent call last):
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/context.py", line 85, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 614, in _doReadOrWrite
    why = selectable.doRead()
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/tcp.py", line 243, in doRead
    return self._dataReceived(data)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/tcp.py", line 249, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 685, in dataReceived
    self._handler()
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 694, in _handleExpected
    self._expected_handler(block, *self._expected_args, **self._expected_kwargs)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 255, in _handleVNCAuth
    self.vncRequestPassword()
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/client.py", line 375, in vncRequestPassword
    self.sendPassword(self.factory.password)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 261, in sendPassword
    des = RFBDes(pw)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/pyDes.py", line 410, in __init__
    self.setKey(key)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 835, in setKey
    if bsrc & (1 << i):
builtins.TypeError: unsupported operand type(s) for &: 'str' and 'int'

Unhandled Error
Traceback (most recent call last):
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/context.py", line 85, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 614, in _doReadOrWrite
    why = selectable.doRead()
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/tcp.py", line 243, in doRead
    return self._dataReceived(data)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/tcp.py", line 249, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 685, in dataReceived
    self._handler()
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 694, in _handleExpected
    self._expected_handler(block, *self._expected_args, **self._expected_kwargs)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 255, in _handleVNCAuth
    self.vncRequestPassword()
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/client.py", line 375, in vncRequestPassword
    self.sendPassword(self.factory.password)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 261, in sendPassword
    des = RFBDes(pw)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/pyDes.py", line 410, in __init__
    self.setKey(key)
  File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 835, in setKey
    if bsrc & (1 << i):
builtins.TypeError: unsupported operand type(s) for &: 'str' and 'int'

Timeout to server localhost:1
`

joukos, 2021-02-01:
Ah, this is more familiar - it's actually an issue in vncdotool and something like this happened:

- issue was reported with a PR that was merged https://github.com/joukos/PaperTTY/pull/60
- problem was supposedly fixed upstream https://github.com/sibson/vncdotool/issues/172
- based on that assumption (and perhaps problem not surfacing for everyone) PaperTTY now specifies `^1.0.0` https://github.com/joukos/PaperTTY/blob/master/pyproject.toml#L16
- turns out it's still broken for some cases at least (tightvnc?) and I think I actually had this same problem fairly recently but didn't bother debugging too much

You might want to try reverting the vncdotool to version `0.13.0` (https://github.com/joukos/PaperTTY/pull/60/files) by editing `pyproject.toml`, installing that and trying again.

thedogbeat, 2021-02-18:
Didn't work for me.
It all ends with `Timeout to server localhost:1`

If it's a vnc issue, do you think this is an okay `netstat -tlpn` response?

`Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:5900            0.0.0.0:*               LISTEN      493/vncserver-x11-c 
tcp        0      0 0.0.0.0:5901            0.0.0.0:*               LISTEN      2188/Xvnc-core      
tcp        0      0 0.0.0.0:6001            0.0.0.0:*               LISTEN      2188/Xvnc-core      
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      399/cupsd           
tcp6       0      0 :::5900                 :::*                    LISTEN      493/vncserver-x11-c 
tcp6       0      0 :::5901                 :::*                    LISTEN      2188/Xvnc-core      
tcp6       0      0 :::6001                 :::*                    LISTEN      2188/Xvnc-core      
tcp6       0      0 ::1:631                 :::*                    LISTEN      399/cupsd       `

Is there another way of pointing to vnc localhost from papertty, apart from `vnc --display 1`?

PS I even bought another Raspberry 3 instead of my 4, but it was stupid I guess :)
Just need some step-by-step instruction or smth.

joukos, 2021-02-22:
I'll just reformat that a bit to make it easier to read:

```
Active Internet connections (only servers) Proto Recv-Q Send-Q Local Address Foreign Address State PID/Program name
tcp 0 0 0.0.0.0:5900 0.0.0.0:* LISTEN 493/vncserver-x11-c
tcp 0 0 0.0.0.0:5901 0.0.0.0:* LISTEN 2188/Xvnc-core
tcp 0 0 0.0.0.0:6001 0.0.0.0:* LISTEN 2188/Xvnc-core
tcp 0 0 127.0.0.1:631 0.0.0.0:* LISTEN 399/cupsd
tcp6 0 0 :::5900 :::* LISTEN 493/vncserver-x11-c
tcp6 0 0 :::5901 :::* LISTEN 2188/Xvnc-core
tcp6 0 0 :::6001 :::* LISTEN 2188/Xvnc-core
tcp6 0 0 ::1:631 :::* LISTEN 399/cupsd
```
There seems to be two displays here (5900 and 5901) - 5900 is for the default X desktop and 5901 is a separate one? It doesn't look like you're running tightvncserver here. That would look like this:

```
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:5901            0.0.0.0:*               LISTEN      1354/Xtightvnc
tcp        0      0 0.0.0.0:6001            0.0.0.0:*               LISTEN      1354/Xtightvnc
```
Did you try with tightvncserver after downgrading vncdotool?

thedogbeat, 2021-04-10:
Just got back to this, downgraded vncdotool, and tried again. Still have an error, but this is new I think.

papertty --driver it8951 vnc --display 0 --password **** --sleep 0.1 --rotate 90
<p>Loading PIL font /home/pi/.local/lib/python3.7/site-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
<p>width = 1200
<br>height = 825
<br>img_addr = 0011a1e0
<br>firmware = SWv_0.2.1T
<br>lut = M841
<br>VCOM = -2.00V
<br>Traceback (most recent call last):
 <br> File "/home/pi/.local/bin/papertty", line 10, in <module>
    sys.exit(papertty.cli())
  <br>File "/home/pi/.local/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
<br>  File "/home/pi/.local/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
<br>  File "/home/pi/.local/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
<br>  File "/home/pi/.local/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
<br>  File "/home/pi/.local/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
<br>  File "/home/pi/.local/lib/python3.7/site-packages/click/decorators.py", line 33, in new_func
<br>    return f(get_current_context().obj, *args, **kwargs)
<br>  File "/home/pi/.local/lib/python3.7/site-packages/papertty/papertty.py", line 583, in vnc
<br>    ptty.showvnc(host, display, password, int(rotate) if rotate else None, invert, sleep, fullevery)
<br>  File "/home/pi/.local/lib/python3.7/site-packages/papertty/papertty.py", line 301, in showvnc
 <br>   client.refreshScreen()
<br>  File "/home/pi/.local/lib/python3.7/site-packages/vncdotool/api.py", line 108, in proxy_call
<br>    raise VNCDoException(result)
vncdotool.api.VNCDoException: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionRefusedError'>: Connection was refused by other side: 111: Connection refused.

Also, this is a `netstat -tlpn` report, it looks different from yours, not sure how it happened.

`Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:6001            0.0.0.0:*               LISTEN      929/Xtightvnc       
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      -                   
tcp        0      0 127.0.0.1:5901          0.0.0.0:*               LISTEN      929/Xtightvnc       
tcp6       0      0 ::1:631                 :::*                    LISTEN `

Also, no need to say how thankful I am to you for helping and also sorry for being extremely lame.

joukos, 2021-04-10:
Connection refused essentially means nothing was listening on that port. According to netstat the server is listening on port 5901, ie. VNC display 1 (0 would be 5900), so try what happens if you pass `--display 1` instead.

thedogbeat, 2021-04-11:
Oh, it's the same as below then.
Speaking of this, I downgraded it, so the report says `
vncdotool --version
vncdotool 0.13.0
`
Should I do something else to get rid of this error?

> Sorry for bothering again, I see that everybody get this working, and I'm still having some strange issue.
> Tried to re-install everything a few times, here's what's going on in case of using `tightvncserver`. The last line is the same though. But maybe there's something in it.
> 
> ```
> (papertty_venv) pi@raspberrypi:~ $ papertty --driver it8951 vnc --display 1 --password ******** --sleep 0.1 --rotate 90
> Loading PIL font /home/pi/papertty_venv/lib/python3.7/site-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
> width = 1200
> height = 825
> img_addr = 0011a1e0
> firmware = SWv_0.2.1T
> lut = M841
> VCOM = -2.00V
> Unhandled Error
> Traceback (most recent call last):
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/log.py", line 103, in callWithLogger
>     return callWithContext({"system": lp}, func, *args, **kw)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/log.py", line 86, in callWithContext
>     return context.call({ILogContext: newCtx}, func, *args, **kw)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/context.py", line 122, in callWithContext
>     return self.currentContext().callWithContext(ctx, func, *args, **kw)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/context.py", line 85, in callWithContext
>     return func(*args,**kw)
> --- <exception caught here> ---
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 614, in _doReadOrWrite
>     why = selectable.doRead()
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/tcp.py", line 243, in doRead
>     return self._dataReceived(data)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/tcp.py", line 249, in _dataReceived
>     rval = self.protocol.dataReceived(data)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 685, in dataReceived
>     self._handler()
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 694, in _handleExpected
>     self._expected_handler(block, *self._expected_args, **self._expected_kwargs)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 255, in _handleVNCAuth
>     self.vncRequestPassword()
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/client.py", line 375, in vncRequestPassword
>     self.sendPassword(self.factory.password)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 261, in sendPassword
>     des = RFBDes(pw)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/pyDes.py", line 410, in __init__
>     self.setKey(key)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 835, in setKey
>     if bsrc & (1 << i):
> builtins.TypeError: unsupported operand type(s) for &: 'str' and 'int'
> 
> Unhandled Error
> Traceback (most recent call last):
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/log.py", line 103, in callWithLogger
>     return callWithContext({"system": lp}, func, *args, **kw)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/log.py", line 86, in callWithContext
>     return context.call({ILogContext: newCtx}, func, *args, **kw)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/context.py", line 122, in callWithContext
>     return self.currentContext().callWithContext(ctx, func, *args, **kw)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/python/context.py", line 85, in callWithContext
>     return func(*args,**kw)
> --- <exception caught here> ---
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/posixbase.py", line 614, in _doReadOrWrite
>     why = selectable.doRead()
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/tcp.py", line 243, in doRead
>     return self._dataReceived(data)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/twisted/internet/tcp.py", line 249, in _dataReceived
>     rval = self.protocol.dataReceived(data)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 685, in dataReceived
>     self._handler()
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 694, in _handleExpected
>     self._expected_handler(block, *self._expected_args, **self._expected_kwargs)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 255, in _handleVNCAuth
>     self.vncRequestPassword()
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/client.py", line 375, in vncRequestPassword
>     self.sendPassword(self.factory.password)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 261, in sendPassword
>     des = RFBDes(pw)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/pyDes.py", line 410, in __init__
>     self.setKey(key)
>   File "/home/pi/papertty_venv/lib/python3.7/site-packages/vncdotool/rfb.py", line 835, in setKey
>     if bsrc & (1 << i):
> builtins.TypeError: unsupported operand type(s) for &: 'str' and 'int'
> 
> Timeout to server localhost:1
> ```

joukos, 2021-04-12:
Seeing that version 0.13.0 should not even have line 835 in its rfb.py ([it's](https://github.com/sibson/vncdotool/blob/v0.13.0/vncdotool/rfb.py) 746 lines long), I suspect the virtual environment might actually be using a different version. When you have the `papertty_venv` active, could you verify if `pip show vncdotool` shows that it's 0.13.0?

thedogbeat, 2021-04-14:
Ohhh yes it works!
I was the 1.0.0 version within virtual environment.
And I did downgrade it in poetry (and it didn't work there), but maybe forgot to do that in `papertty_venv`.
So, that was it i guess. It also works with `x11vnc` btw, not only `tightvnc` (they are interchangeable, right?)
Thank you so much @joukos !

arahasya, 2021-04-20:
I had the exact same issues and made it run finally.

I can see the desktop now but the keyboard and mouse aren't working. Any ideas?

arahasya, 2021-04-28:
Help please

joukos, 2021-04-28:
@thedogbeat good to hear!

@arahasya could you give a bit more detail - where's the input coming from, another VNC client or plugged straight to the RPi's USB ports? Are you using x11vnc or a separate display?

arahasya, 2021-05-01:
I have followed all your recommendations from above. Using tightvncserver and changed version of vncdotool
display is 9.7 waveshare IT8951

joukos, 2021-05-02:
@arahasya it's good that you see the desktop, but note that the above recommendations were not for your use case specifically, so I can't know what you're attempting to do and how, unless you share that information.

The reason I ask is to establish what your particular use case is and whether you're using VNC in the right way for it. Do I understand correctly that you have tightvncserver running a VNC display, shown on the e-ink via PaperTTY, with mouse and keyboard connected to the same RPi as the display?

If that is the case, that will **not** work (without something extra). The reason is that the VNC display has no access to the connected peripherals nor an easy way to connect them to it (as far as I know). One way is covered in the main README (search for "Standalone RPi with mouse and keyboard"), but it requires some extra tweaking to be nice.

There's also another, arguably perhaps better way but undocumented at the moment, of using a virtual screen for X that would be the VNC display. I haven't actually tried it being the *only* display, but have used it with a laptop so that:

- laptop is configured to have a virtual X screen (say, to the left of the main display) that's shared with x11vnc
- RPi runs PaperTTY that connects to laptop's x11vnc display
- moving mouse from laptop's main screen to the left into the virtual screen now displays it on the e-ink

I'm not sure right now if such virtual screens can be configured similarly for a standalone RPi, but if there's nothing stopping that, I think the cleanest way would be to configure X to use a single virtual screen and run x11vnc for that, then run PaperTTY to connect to that one. There's some scripts that might help in this (https://github.com/mrenrich84/vnc_virtual_display_linker) but last time I tried this I just wrote my own that fit my use case better.

If this is something you're interested in, I can see if I find those scripts. I was planning to add a section to documentation about this "feature" but as always, the doc overhaul is still waiting for me to have time to do it.

joukos, 2021-05-08:
Reason for having VNC in the mix is mostly about it being flexible and "headless" so it could be used for as many purposes as possible, and also because it was fairly trivial to implement and I wanted to get quick results. It's certainly not optimal for the standalone use case, but I haven't had time to work out the details with regard to frame buffer, since that also has some kinks (like how to handle a resolution too big for the display, should there be an option for scrolling etc. and how to control that). For the same lack-of-time reason README is also pending an update, sorry for it being messy.

PR for implementing such frame buffer copy is very welcome, of course.

PompTart, 2022-01-15:
I'm having the same issue when trying to follow these steps:
In one terminal window, tmux to session 0:
# tmux window/pane #0
sudo lightdm
export $(sudo x11vnc -findauth)
sudo x11vnc -usepw -display :0 -forever

In another terminal window, tmux to session 1:
sudo papertty --driver epd2in13 vnc --display 0 --password supAPass --sleep 0.1 --rotate 90

Sometimes I get the "unsupported operand" error message, while other times I get the connection refused one. I did downgrade vncdotool to 0.13.0 and reinstalled. What else could it be?

joukos, 2022-01-15:
@PompTart how exactly did you downgrade, does `pip show vncdotool` show the correct version? If you get the `unsupported operand` error, it's almost certain it's using the wrong version when you do `sudo papertty`.

If you get a `connection refused`, that on the other hand means the VNC server wasn't running at the time or the port was wrong.

PompTart, 2022-01-16:
I'm getting "unsupported operand". `pip show vncdotool` shows Version 0.13.0. I downgraded by editing pyproject.toml and reinstalling papertty.

I should also mention that I'm getting other errors and warnings.
`sudo lightdm` ends in a segmentation fault after "Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files"

And then, when I run papertty in the second window, I get a warning "You do not have a working installation of the service_identity module". I tried installing this module but I still get the warning.

PompTart, 2022-01-16:
A little more context: I'm ultimately trying to mirror a terminal window running a text editor (wordgrinder) on the e-paper. I have to use a graphical OS because Plover (stenography software) must be running in the background. When I run papertty in terminal mode, it works pretty well but the refresh/update is quite slow (between 1 and 3 seconds); I don't think it's doing partial refresh. I'm using the newer 6" HD display. I have tried George's fork but still can't figure out the partial refresh (although his graphics demo does seem to work). I'm using the Pi Zero 2 W and also have it connected to an HDMI display, but hope to use the e-paper alone.

joukos, 2022-01-17:
How did you (re)install papertty, using pip or poetry? Admittedly the docs are still woeful regarding the basic setup after poetry/pip were introduced, but here's some general possibilities wrt installing:
1. With pip globally
   - Messy
   - `pip3 install papertty`
   - `papertty`
1. With pip for current user
   - Messy still
   - `pip3 install --user papertty`
   - `papertty` (if PATH is configured properly)
1. With pip in a virtualenv
   - Better, ie. dependencies are isolated in the venv
   - `python3 -m venv venv`
   - `venv/bin/activate`
   - `pip install papertty`
   - `papertty` (when env is activated)
1. With poetry
   - `poetry install`   
   - Poetry creates a venv implicitly (you can get the path with `poetry env info -p`)
   - `poetry shell` (activates the venv)
   - `papertty`
   - **Or:** `$(poetry env info -p)/bin/papertty` (running this means that the venv will be used)

Depending which way you used downgrading the package may need to be done differently. That is, if you used pip, modifying `pyproject.toml` doesn't really do much, but `poetry install` (or `poetry update` in this case) would pick up the change. However to make use of it, you need to run the helper script from within the venv that poetry had created (`$(poetry env info -p)/bin/papertty`). It can be a bit confusing sometimes.

However I now decided to take a quick look at implementing a more conventional framebuffer support so messing with VNC etc. wouldn't be needed, and turns out it was much more trivial than I expected, so you might try that instead. The draft PR is here https://github.com/joukos/PaperTTY/pull/86, but to try it out, you can:

```bash
# clone the repo somewhere
git clone https://github.com/joukos/PaperTTY.git
cd PaperTTY
git checkout fb-proto
poetry install
# add --rotate 180 or such if needed
$(poetry env info -p)/bin/papertty --driver IT8951 fb --invert --sleep 0.01
```

This would draw the /dev/fb0 to the e-ink directly, so you should see the console or GUI there (will be rescaled to fix the display). Please let me know if it works, as this was just a quick hack and "works for me"  :slightly_smiling_face: . In order to make things readable you may want to set some high-contract theme without a background image and big enough fonts in black/white.

Here's just a quick pic with the 4.2" showing LXDE:
[image]

joukos, 2022-01-17:
P.S. I merged this change to `master` now (will need to some day rename to `main`...) even if it's experimental, as this is a highly desired feature. Will push the pip package too in a bit.

PompTart, 2022-01-19:
I installed with `pip install papertty` but I'll try the more conventional framebuffer way you suggested.

PompTart, 2022-01-19:
Okay when I ran `poetry install` it installed all the packages starting at attrs and all the way down to zope.interface, but failed at pillow (7.1.2), followed by a very lengthy error message, which is hard to summarize, but startes with EnvCommandError

joukos, 2022-01-19:
Maybe try `sudo apt install libjpeg-dev` - I also got some `jpeg` related error prior (and I think it maybe used to be a dependency of some other package because I don't recall needing to install it specifically).

PompTart, 2022-01-21:
I ran a 'sudo apt-get update' and 'sudo apt install libjpeg-dev'. Pillow and a few more packages installed fine. Now the rpi.gpio package is giving an EnvCommandError.

joukos, 2022-01-21:
@PompTart would you paste the full error you see? Also can you elaborate which Raspberry Pi you're using, which OS version and also the `python --version` that's installed?

PompTart, 2022-01-21:
It's the Pi Zero 2 W, running the desktop OS that I installed with their installer. I ran the recommended update.
The `python --version` is 3.9.2.
Here's the message:
pi@raspberrypi:~/PaperTTY $ poetry install
Installing dependencies from lock file

Package operations: 1 install, 0 updates, 0 removals

  • Installing rpi.gpio (0.7.0): Failed

  EnvCommandError

  Command ['/home/pi/.cache/pypoetry/virtualenvs/papertty-_ljrO54G-py3.9/bin/pip', 'install', '--no-deps', 'file:///home/pi/.cache/pypoetry/artifacts/d0/a5/a6/f41ed6a446a382fe3ee625ab60f88dc3d49564b51a0c4538c23ca9c0c4/RPi.GPIO-0.7.0.tar.gz'] errored with the following return code 1, and output: 
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
  Processing /home/pi/.cache/pypoetry/artifacts/d0/a5/a6/f41ed6a446a382fe3ee625ab60f88dc3d49564b51a0c4538c23ca9c0c4/RPi.GPIO-0.7.0.tar.gz
    Preparing metadata (setup.py): started
    Preparing metadata (setup.py): finished with status 'done'
  Building wheels for collected packages: RPi.GPIO
    Building wheel for RPi.GPIO (setup.py): started
    Building wheel for RPi.GPIO (setup.py): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.cache/pypoetry/virtualenvs/papertty-_ljrO54G-py3.9/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-8n4fgm5d/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-8n4fgm5d/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-_bt_ho22
         cwd: /tmp/pip-req-build-8n4fgm5d/
    Complete output (87 lines):
    running bdist_wheel
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.9
    creating build/lib.linux-armv7l-3.9/RPi
    copying RPi/__init__.py -> build/lib.linux-armv7l-3.9/RPi
    creating build/lib.linux-armv7l-3.9/RPi/GPIO
    copying RPi/GPIO/__init__.py -> build/lib.linux-armv7l-3.9/RPi/GPIO
    running build_ext
    building 'RPi._GPIO' extension
    creating build/temp.linux-armv7l-3.9
    creating build/temp.linux-armv7l-3.9/source
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/.cache/pypoetry/virtualenvs/papertty-_ljrO54G-py3.9/include -I/usr/include/python3.9 -c source/c_gpio.c -o build/temp.linux-armv7l-3.9/source/c_gpio.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/.cache/pypoetry/virtualenvs/papertty-_ljrO54G-py3.9/include -I/usr/include/python3.9 -c source/common.c -o build/temp.linux-armv7l-3.9/source/common.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/.cache/pypoetry/virtualenvs/papertty-_ljrO54G-py3.9/include -I/usr/include/python3.9 -c source/constants.c -o build/temp.linux-armv7l-3.9/source/constants.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/.cache/pypoetry/virtualenvs/papertty-_ljrO54G-py3.9/include -I/usr/include/python3.9 -c source/cpuinfo.c -o build/temp.linux-armv7l-3.9/source/cpuinfo.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/.cache/pypoetry/virtualenvs/papertty-_ljrO54G-py3.9/include -I/usr/include/python3.9 -c source/event_gpio.c -o build/temp.linux-armv7l-3.9/source/event_gpio.o
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/.cache/pypoetry/virtualenvs/papertty-_ljrO54G-py3.9/include -I/usr/include/python3.9 -c source/py_gpio.c -o build/temp.linux-armv7l-3.9/source/py_gpio.o
    source/py_gpio.c: In function ‘PyInit__GPIO’:
[... cut ...]  Check the logs for full command output.
  

  at ~/.poetry/lib/poetry/utils/env.py:1183 in _run
      1179│                 output = subprocess.check_output(
      1180│                     cmd, stderr=subprocess.STDOUT, **kwargs
      1181│                 )
      1182│         except CalledProcessError as e:
    → 1183│             raise EnvCommandError(e, input=input_)
      1184│ 
      1185│         return decode(output)
      1186│ 
      1187│     def execute(self, bin, *args, **kwargs):

joukos, 2022-01-22:
Ah, okay, I tried this now with a fresh Bullseye image and I get the same, I'm guessing it's related to the GPIO lib not being compatible with Python 3.9. I'll see if I have time today to work around it, perhaps just installing Python 3.8 and telling poetry to use that instead would work.

joukos, 2022-01-22:
@PompTart here's a workaround, apparently the prerelease version works with 3.9: https://github.com/joukos/PaperTTY/issues/87

PompTart, 2022-01-23:
okay everything seems to have installed properly by changing `pyproject.toml` as you specified.
Now, how do you get this beginning part to appear in the terminal? 
`$(poetry env info -p)/bin/papertty --driver IT8951 fb --invert --sleep 0.01`
I assume that not all of that is typed in as input... Was that header part supposed to appear after running `poetry update` or `poetry install`?

PompTart, 2022-01-23:
Ope, never mind, I ran it with the `env` commands that you posted earlier. Okay, I'm seeing the contents of tty1 on the e-paper display. I'm going to try to get one of the terminal windows from the graphical OS to appear on the e-paper. As I said, it needs to be that way in order for Plover to accept stenography input from my keyboard.

PompTart, 2022-01-23:
This is the third OS I've installed on this machine, the other two being dietpi and Pi Os Lite, and now I'm having difficulty running Plover. I hope I get some combination of programs and OSes to work. All I need is a GUI terminal window mirrored on the e-paper with partial refresh.
The partial refresh does seem to be working with this version of papertty in the envelope.

PompTart, 2022-01-23:
The partial refresh is faster than the total refresh, but it seems like two different things are going on.
1. The area that was recently edited is refreshed 
2. About a quarter of the screen containing that area is refreshed.

joukos, 2022-01-23:
The partial update redraws the whole area of the bounding box of differences between frames, so if for example 1 pixel changes at top-left and 1 pixel at bottom-right, the whole display area would need to refresh anyway. Also if for example the cursor moves, we need to update the spot where it disappeared from and where it landed, so this would also update a rectangle that contains both changes.

This is not "optimal" in terms of minimizing the amount of image data that is sent to the display, but doing several smaller updates in sequence would likely be faster only in pathological cases.

If the mouse is not moving and especially if there's no widgets updating in corners of the screen (clock, updating graphs, focus changing colors etc.), there shouldn't be anything much to update, so getting it to play nice may also involve a bit of UI tweaking.

Note also that there's some silly bugs https://github.com/joukos/PaperTTY/blob/master/papertty/papertty.py#L341 because I forgot for a while how everything worked and currently a full refresh is _not_ actually made every N frames as expected. I fiddled with this a bit yesterday and there's quite a few bugs pending fix and possibly some optimizations, but maybe if I get my 6" working again it's easier to test these...

joukos, 2022-01-23:
What was the problem with Plover, btw? I checked it out briefly with `sudo apt install plover` (installs version `4.0.0~dev8~66~g685bd33-2`) and it seemed to start up at least. Didn't test it for real yet, though.