joukos/PaperTTY #74: Mirror function for VNC

serenewaffles, 2021-01-31:
I have a 10.3" 1872x1404 display. I'm able to get VNC working on it great, but I can't get VNC to mirror the image, so all the text is backwards.

joukos, 2021-01-31:
Oh, that's a bit unexpected, but should be easy to fix - currently the different functions have a slightly differing set of image operations supported, so VNC just doesn't happen to have mirroring at the moment as an option.

If you take a look at the image display functionality (https://github.com/joukos/PaperTTY/blob/master/papertty/papertty.py#L453-L456) it optionally does mirroring for the image:

```python
    if mirror:
        image = ImageOps.mirror(image)
    if flip:
        image = ImageOps.flip(image)
```

I'm guessing you just need to do `ImageOps.mirror()` for the to-be-drawn image, so before creating a PR (that maybe makes the VNC feature use the same `display_image()` instead of just adding a mirroring option to avoid duplication), you could try to modify this part:

https://github.com/joukos/PaperTTY/blob/master/papertty/papertty.py#L313

```python
                # apply invert
                if invert:
                    new_vnc_image = ImageOps.invert(new_vnc_image)
                # rescale image if needed
                if new_vnc_image.size != (self.driver.width, self.driver.height):
                    new_vnc_image = new_vnc_image.resize((self.driver.width, self.driver.height))
```

To be:

```python
                # apply invert
                if invert:
                    new_vnc_image = ImageOps.invert(new_vnc_image)
                new_vnc_image = ImageOps.mirror(new_vnc_image)
                # rescale image if needed
                if new_vnc_image.size != (self.driver.width, self.driver.height):
                    new_vnc_image = new_vnc_image.resize((self.driver.width, self.driver.height))
```

Just to see if that fixes the problem.

serenewaffles, 2021-01-31:
Yes! That fixed my problem! I think it has something to do with my panel, because even using the Waveshare demo code, everything appears mirrored for me.

joukos, 2021-01-31:
Good to hear. Not sure why your unit would be "backwards" though, but adding a mirroring option isn't a lot of work. PRs are welcome, but I might get around to doing it at some point too, but likely not in the next few days.