joukos/PaperTTY #75: Sudo with papertty?

Adam90a, 2021-02-15:
I am stuck. I try to use sudo and get an error...

sudo papertty --driver EPD4IN2 terminal
sudo: papertty: command not found

...and without....

papertty --driver EPD4IN2 terminal
Loading PIL font /home/pi/.local/lib/python3.7/site-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
No read access to /dev/vcsa1 - maybe run with sudo?

.....any suggestions?

closed 2021-02-16 by Adam90a

Julian-MJK, 2021-10-04:
Dear god if you fixed this, tell us how, don't just close the damn issue so that nobody who has the same issue in the future can figure it out

Mozartenhimer, 2021-10-04:
This looks like a cli usage thing. It's likely that '. (current directory)
is not on the path for root,  it is for the user invoking sudo. The
solution for running it under sudo on this instance would be:
sudo  ./papertty

gridbasedotnet, 2023-08-12:
Did you ever get this resolved?

Adam90a, 2023-08-12:
Nope, gave up