joukos/PaperTTY #77: Add documentation on pypy use

a8ksh4, 2021-03-23:
Just though I'd share my notes here on how to use pypy with papertty.  I'm working on a 4in2 display with a pi zero and am trying to improve performance.  Not sure yet if pypy helps, but here's a short how-to:

```
sudo apt install pypy3 pypy3-devel libjpeg-dev zlib1g-dev wiringpi virtualenv
virtualenv --python=/usr/bin/pypy3 papertty_venv_pypy
. papertty_venv_pypy/bin/activate
pip3 install papertty
```

For fonts:
`sudo apt install ttf-mscorefonts-installer`
and to see what's provided:
`ls /usr/share/fonts/truetype/msttcorefonts`

And in my start.sh, I use:
```
. /home/pi/papertty_venv_pypy/bin/activate
papertty --driver EPD4in2 terminal --size 11 --autofit --font /usr/share/fonts/truetype/msttcorefonts/cour.ttf --portrait
```

Also, thanks everyone who has contributed to this tool.  This is super cool stuff. :D

joukos, 2021-04-10:
Sorry, I had completely missed this issue somehow and only now noticed it. Thanks, and please share here any findings you may have. I don't know if it would improve the overall performance much (the most significant avenue for optimization for the larger displays at least might be the SPI transfers) but even small gains are good to know about, especially since the use cases vary quite a bit and for the lower end RPis the effect might be more pronounced.

a8ksh4, 2021-04-19:
Hi Joukos,  I don't think it made much difference in performance of the display drivers.  I ended up just moving to a pi 4 and things are much faster. :)  I'm using ubuntu now and haven't gotten pypy working there yet.  Will update here if I figure it out. 

It seems odd to me that even a partial refresh, like one character updated, kicks the display process up to 90% CPU even for a second.  I hope I'll have time to look closer at the drivers and see how they work, but won't hold my breath. :P 

Here's a little linux console handheld I put together using PaperTTY and a 4.2" screen:

[image]

joukos, 2021-04-20:
@a8ksh4 that's beautiful! Thanks for sharing :)

assadollahi, 2021-04-20:
love the design, do you have more pictures or a youtube video?

a8ksh4, 2021-04-23:
@assadollahi No youtube, but I put some info up on gitlab: https://gitlab.com/norris.daniel/paper-pi-handheld/-/tree/master

closed 2022-01-18 by joukos