joukos/PaperTTY #80: How to install / run PaperTTY?

DieBe, 2021-07-11:
Maybe it's obvious and I'm just being very blind, but as a very inexperienced user, I don't know how to install PaperTTY. 

`sudo apt install python3-venv python3-pip libopenjp2-7 libtiff5`
`python3 -m venv papertty_venv`
`source papertty_venv/bin/activate`
`pip3 install papertty`

Is running this enough? I'm using the Waveshare 13.3" display, it's using the IT8951 so I'm hoping this might acutally work, even though it's not officially supported. 
When I runthis, everything seems to be installed correctly, but the screen doesn't do anything. Do I have to runa startup file or something like that?

Sorry for my propably unnecessary question.
Thanks for any help,
DieBe

DieBe, 2021-07-11:
A little update:
I finally fount the passage in the readme. Now I'm at the point where I should select the driver, in my case the IT8951. But when running `papertty --driver IT8951`, it says: 
Usage: `papertty [OPTIONS] COMMAND [ARGS]...`
As mentioned, I'm really inexperienced in Linux.
One more thing: In the readme, it always says `sudo papertty` + command. When I'm running pretty much anything with `sudo` before the `papertty`, it says Command not fount.

joukos, 2021-07-12:
Sorry about the confusing state of the docs, I still haven't had the time/energy to overhaul them.

You might also try adding `--help` to the command to see perhaps a bit more info, but have you tried something like `echo hello | papertty --driver it8951 stdin` (doc [here](https://github.com/joukos/PaperTTY#stdin---render-standard-input)), if that shows anything? Might be just a very small squiggle in a corner due to the small default font but if that doesn't complain, things should be fairly good.

The way the command line works in PaperTTY is that there's a few "top-level" options such as the driver selection, and these would come _before_ a "subcommand" (ie. `terminal` / `vnc` / `stdin` / ...) and each of those have their own options (add `--help` to see them, for example `papertty --driver it8951 stdin --help` and it will print out the available options for the `stdin` subcommand. The CLI might be slightly unintuitive to discover and config file support would help.

Regarding `sudo`, reason it can't find the command is that the environment is not the same (eg. `$PATH` and friends - you can see this if you have the virtualenv activated and run `env` and `sudo env`) but you can work around this by running `sudo papertty_venv/bin/papertty ...`, since the supplied script invokes the program via Python interpreter within the virtualenv. `sudo` is not strictly necessary, but unless you set up permissions otherwise to the necessary devices (such as `/dev/vcs*`) you might need it in some cases, such as terminal mode. This is another thing that could be improved, though.

Please let me know if this helps or if it doesn't, interested to hear if the 13.3" works.

DieBe, 2021-07-13:
Thanks for your answer. I just made it working, turned out I was pretty blind while reading through the readme. It works with the 13.3" too, which is pretty amazing. The only problem is the refresh rate, it only updates every 5 seconds, so if I input something it takes about 5 seconds to even start the screen refresh. Is there any way to speed this up? I'm using it with X11VNC.

joukos, 2021-07-13:
Good to hear it works!

Which RPi are you using it with? The lower end ones will be pretty slow, something like RPi4 should be a lot faster dealing with any overhead, but in the end the SPI transfer will take majority of the time. You can however try to tweak this https://github.com/joukos/PaperTTY/blob/master/papertty/drivers/driver_it8951.py#L183 to, say, `16000000` if your RPi is beefier (lower it a bit if things don't work anymore). There were some other places that can be optimized, but the real fix for making it faster would probably be to rewrite the SPI stuff altogether. Here I mention changing some places to use `writebytes2` - unfortunately I don't have that branch handy at the moment, but those were very minor changes anyway - https://github.com/joukos/PaperTTY/issues/32#issuecomment-752866792, and it updated reasonably fast.

To make things as fast as possible, using B/W instead of grayscale should be faster, but there's not a nice switch for that either at the moment, but is somewhat described in another comment here https://github.com/joukos/PaperTTY/issues/32#issuecomment-753225886.

DieBe, 2021-07-13:
I'm using a  Pi 3 A+, since I'll build an e-reader out of it. (Less tall)
I'll try the stuff you mentioned lateron today.

DieBe, 2021-07-13:
Setting the maximum frequency to 20000000 didn't really change anything... Even though it's 10x than the standard setting more. Setting ist to 12000000 or 16000000 neither did. Are you sure this is the limiting parameter?

joukos, 2021-07-13:
Hmm, did you make the edit in the "pip installed" files, ie. in `papertty_venv/lib/python3.7/site-packages/papertty/drivers/driver_it8951.py`?

serafimdasaudade, 2021-07-13:
Hi, 
I'm got waveshare 6inch with the rpi hat with the it8951 driver, and I'm also having performance issues.
I'm using rpi 4. Change the spi speed from  2000000 to 24000000 and a didn't see any changes. 

Cumps,

DieBe, 2021-07-17:
So far, everything is working with the 13.3" display. But it is really slow. Big updates take at least 5 seconds, sometimes even longer. For my use, a black and white mode would be perflectly fine if it speeds up the data transfer over SPI (which seems to be the bottleneck). However, editing papertty.py (in papertty_master or papertty_ven) doesn't change anything. Even after renaming those files it still works as before. Which files do I have to edit? Or is there another way to edit the image before it's being sent do IT8951? Are there other options to speed it up?

Molerat89, 2021-07-18:
Hello!

I got the waveshare eink 6" display and I cannot run it with vnc. I did a clean setup of the Pi from scratch, just to make sure it's not interacting with other modules. I'm using a Pi4 4GB with the latest raspbian installed. I followed the installation from front page:

$ sudo apt install python3-venv python3-pip libopenjp2-7 libtiff5
$ pip3 install papertty

I wanted to make sure the driver works, so I tried the following command:

$ sudo papertty --driver IT8951 image --image "/home/pi/Downloads/test.jpg"

It worked fine.

I'm using realvnc server and I typed the following to test it:

$ vncserver -geometry 400x300 :1
$ sudo papertty --driver IT8951 vnc --display 1 --password supAPass --sleep 0.1 --rotate 90

Got a flash on the screen and after a while this:

"Timeout to server localhost:1"

Tried with different password (raspberry), with no password, with tightvnc (which gives a lot of errors back when running papertty). The vnc is running because I can see it on my pc side and I can connect to it with 192.168.1.9:1.
I've been trying to run it for 2 days in a raw and still cannot understand what's wrong. I'm assuming this is an authentication issue (?). 

Thank you in advance, any help will be really appreciated

joukos, 2021-07-20:
@DieBe probably forcing the B/W mode would help, just need to hack the code a bit to do that. Also using `writebytes2` where applicable should have some impact, but I don't have the branch handy to check the necessary changes (there's not many).

@Molerat89 you may want to check out https://github.com/joukos/PaperTTY/issues/72 and if that applies to your case. RealVNC seems to have issues there and I've only tested with tightvncserver.

Molerat89, 2021-07-21:
@joukos 
Thank you for your guidance on this particular issue.
It still doesn't work for me though.

So far I have downgraded vncdotool to version 0.13.0 in virtualenv, pip and pip3.
Also I installed tightvnc and run it. The password is set to raspberry and I'm getting this:

`$ sudo papertty --driver IT8951 vnc --display 2 --password raspberry --sleep 0.1 --rotate 90
Loading PIL font /usr/local/lib/python3.7/dist-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
width = 1448
height = 1072
img_addr = 00122480
firmware = SWv_0.6.
lut = M841_TFAB512
VCOM = -2.00V
Unhandled Error
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/twisted/python/log.py", line 101, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/local/lib/python3.7/dist-packages/twisted/python/log.py", line 85, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/local/lib/python3.7/dist-packages/twisted/python/context.py", line 118, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/local/lib/python3.7/dist-packages/twisted/python/context.py", line 83, in callWithContext
    return func(*args, **kw)
--- <exception caught here> ---
  File "/usr/local/lib/python3.7/dist-packages/twisted/internet/posixbase.py", line 687, in _doReadOrWrite
    why = selectable.doRead()
  File "/usr/local/lib/python3.7/dist-packages/twisted/internet/tcp.py", line 246, in doRead
    return self._dataReceived(data)
  File "/usr/local/lib/python3.7/dist-packages/twisted/internet/tcp.py", line 251, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 685, in dataReceived
    self._handler()
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 694, in _handleExpected
    self._expected_handler(block, *self._expected_args, **self._expected_kwargs)
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 255, in _handleVNCAuth
    self.vncRequestPassword()
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/client.py", line 375, in vncRequestPassword
    self.sendPassword(self.factory.password)
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 261, in sendPassword
    des = RFBDes(pw)
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/pyDes.py", line 410, in __init__
    self.setKey(key)
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 835, in setKey
    if bsrc & (1 << i):
builtins.TypeError: unsupported operand type(s) for &: 'str' and 'int'

Unhandled Error
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/twisted/python/log.py", line 101, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/local/lib/python3.7/dist-packages/twisted/python/log.py", line 85, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/local/lib/python3.7/dist-packages/twisted/python/context.py", line 118, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/local/lib/python3.7/dist-packages/twisted/python/context.py", line 83, in callWithContext
    return func(*args, **kw)
--- <exception caught here> ---
  File "/usr/local/lib/python3.7/dist-packages/twisted/internet/posixbase.py", line 687, in _doReadOrWrite
    why = selectable.doRead()
  File "/usr/local/lib/python3.7/dist-packages/twisted/internet/tcp.py", line 246, in doRead
    return self._dataReceived(data)
  File "/usr/local/lib/python3.7/dist-packages/twisted/internet/tcp.py", line 251, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 685, in dataReceived
    self._handler()
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 694, in _handleExpected
    self._expected_handler(block, *self._expected_args, **self._expected_kwargs)
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 255, in _handleVNCAuth
    self.vncRequestPassword()
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/client.py", line 375, in vncRequestPassword
    self.sendPassword(self.factory.password)
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 261, in sendPassword
    des = RFBDes(pw)
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/pyDes.py", line 410, in __init__
    self.setKey(key)
  File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 835, in setKey
    if bsrc & (1 << i):
builtins.TypeError: unsupported operand type(s) for &: 'str' and 'int'

Timeout to server localhost:2
`
I'm fairly new to python, so I saw something about editing the requirements.txt on #72 ,but I don't know the location of it and how to edit it.

Ps= When I reinstalled vncdotool version 0.13.0, I got a message that papertty is not supporting it and it needs version 1.0.0
Ps2= `$ tightvncserver -geometry 400x300 :2` is the command I'm running to start the vnc. I tried already other displays but no luck so far. 

Thank you for your suggestions!

x20011, 2021-07-22:
Hello there,
since I have the same issue, would you explain how you managed to change vncdotool v1.0.0 to v0.13.0?

Thank you

joukos, 2021-09-26:
Sorry for the late answer (was on vacation etc. and missed this).

```
File "/usr/local/lib/python3.7/dist-packages/vncdotool/rfb.py", line 835, in setKey
```

This is a dead giveaway that you're not actually using the 0.13.0 with the PaperTTY that runs, ie. if you check https://github.com/sibson/vncdotool/blob/v0.13.0/vncdotool/rfb.py , there is no line 835 in that version. Similar issue was here https://github.com/joukos/PaperTTY/issues/72#issuecomment-817561156 so you might want to check that the virtualenv you run it in has the correct version, and if you tell pip to install a particular version, it must be the pip that's inside that environment.

joukos, 2022-01-18:
I assume this has been resolved, also the README has been updated slightly with regard to installation steps.

closed 2022-01-18 by joukos