joukos/PaperTTY #83: TTY refused to resize with "[Errno 22] Invalid argument"

hym3242, 2021-12-10:
- raspberry pi os (10 buster) 
- rpi zero w
- waveshare 2.13inch e-Paper HAT V2
- installed papertty from pypi

Running 
`sudo papertty --driver EPD2in13v2 terminal --autofit`
results in 
`Automatic resize of TTY to 20 rows, 62 columns`
`TTY refused to resize (rows=20, cols=62), continuing anyway.`
however using another newly burnt sd card i am able to resize the tty.

with further investigations i found that the ioctl for resizing the tty is producing this error
`[Errno 22] Invalid argument`
with this ioctl (however the attempted ioctls are identical on both cards)
`fcntl.ioctl(8,21524,b'\x14\x00>\x00\x00\x00\x00\x00')`

I really wonder why this is happening. How could the same ioctl fail on one system and success on another?
Or maybe suggest a way to manually resize the tty once inside?

hym3242, 2021-12-10:
strace shows me this `ioctl(8, TIOCSWINSZ, {ws_row=20, ws_col=62, ws_xpixel=0, ws_ypixel=0}) = -1 EINVAL (Invalid argument)`

joukos, 2022-01-02:
Hi, sorry about late reply.

Both are running the same buster I presume but do they have the same kernel versions? The TTY resize is pretty finicky in general unfortunately.

hym3242, 2022-01-09:
> Hi, sorry about late reply.
> 
> Both are running the same buster I presume but do they have the same kernel versions? The TTY resize is pretty finicky in general unfortunately.

No they are not running the same kernel. 
The working one is 5.10.17+
The not working one is  4.19.118+

hym3242, 2022-01-15:
After some experimenting I think I have found the culprit.
If anyone else is facing this error, try increasing the tty1 terminal size... it seems the tty doesn't want to be enlarged but only scaled down.

joukos, 2022-01-18:
Thanks for the info. Implementing a proper pseudo TTY support would help here, but that's still pending. I'll close the issue for now.

closed 2022-01-18 by joukos