joukos/PaperTTY #88: Stuck in loading

js22gz, 2022-02-18:
Hello and thank you for creating PaperTTY!
This might not be a Issue with the code but an issue with my lack of skill in this area, but here's the problem: 

I followed the new instructions by installing Poetry. 
From the terminal I do this: 

1. cd PaperTTY
2. poetry shell 
3. papertty --driver EPD7in5b terminal 

The program starts and gets stuck after printing this line (and nothing shows on the screen): 
"Loading PIL font /home/pi/PaperTTY/papertty/resources/tom-thumb.pil. Font size is ignored."
(I've tried all the 7.5 drivers with same result)

I have a Raspberry Pi 3 A+  with a 7.5" Waveshare Red, White, Black ePaper-screen V2. 
Connecting the two through a ePaper Driver HAT Rev2.2
I got the examples in Waveshare git-repo to work. 

I don't really know where to start troubleshooting and would really appreciate being pointed in the right direction.

joukos, 2022-02-20:
If the Waveshare examples work, first thing that comes to mind here is that likely there's currently no working support for 7.5" _tricolor **V2**_ and I don't know if anyone's actually tested the non-V2 versions of these more exotic models either.

The way to troubleshoot this would basically be to get the Python examples from Waveshare, then see how similar they are to the existing PaperTTY drivers (ie. [here](https://github.com/joukos/PaperTTY/blob/master/papertty/drivers/drivers_full.py#L448) and [here](https://github.com/joukos/PaperTTY/blob/master/papertty/drivers/drivers_full.py#L546)) and if it's possible to just add some minor tweaks to get it to work or if it's quite different, in which case it's likely easiest to just adapt the code from the Waveshare examples as a starting point.

It's basically just writing the right bytes to the display so the code shouldn't be super complex, but I don't have this particular model myself so I can't really test it unfortunately. Some displays have a reversed "ready" bit so when nothing seems to happen, it could even be that the code expects a 0 when it should be expecting a 1, and debugger could be helpful there.

js22gz, 2022-02-25:
Thank you for your answers! 
After tinkering with the screen a few hours using the examples from waveshare I cant get the screen to work on PaperTTY and even just using the waveshare examples I cant get the refreshrate to go under 20s... So I think the screen is out of PaperTTYs reach to be honest... 

The problem (I believe) is with the lack of partial refresh and the fact that I cant find where to change the LUT tables to maybe tweak the refreshrate a little further... I should have researched more before hitting the buy-button :-)

joukos, 2022-02-25:
It can be made to work with PaperTTY surely, but in these tricolor models the refresh rate indeed is very slow due to the way it works physically. In some years we might have some better displays but for the time being those are mostly suitable for fairly static information.