joukos/PaperTTY #89: Increase refresh speed - waveshare 6inch HD e-Paper HAT

serafimdasaudade, 2022-02-25:
Hi, 
I'm testing this lib.
I'm using the vnc desktop, and it working. But I start notice that e-paper take a lot of time to update +- 5sec.
The command I'm using:
`poetry run papertty --driver it8951 vnc --display 1 --password raspberry --sleep 0.1 --rotate 0`

I would like to improve the refresh rate. Any tips?

Cumps,

joukos, 2022-02-25:
Which RPi are you using? To get decent speeds with a display this size an RPi3b+ / RPi4 might be best. There's a lot of room for optimization in general, though.

serafimdasaudade, 2022-02-25:
I'm using a RPI 4.
I'm doing some code teaks, but with no notice improvements.

serafimdasaudade, 2022-03-23:
> Which RPi are you using? To get decent speeds with a display this size an RPi3b+ / RPi4 might be best. There's a lot of room for optimization in general, though.

Hi, Do you got some tips. Help me go to the correct direction?
 

Cumps,

chi-lambda, 2022-04-01:
You can set the `max_speed_hz` ([driver_it8951.py:183](https://github.com/joukos/PaperTTY/blob/3ea8286903b98fac071285008b4cc05dd84c2121/papertty/drivers/driver_it8951.py#183)) to something higher. 18 MHz worked for me, but it stopped responding after running it for a few hours. Could be unrelated though. There was no permanent damage as far as I can tell. VNC uses the grayscale instead of black-and-white mode, which uses more data (I think it was a factor of four because even b/w uses 2 bits per pixel) and is is thus slower. You could add another filtering step to turn it into a black-and-white picture after [papertty.py:369](https://github.com/joukos/PaperTTY/blob/cf469547cd7f77dd97633c3e006b29830e121acd/papertty/papertty.py#L369) to speed things up if you can accept the loss of quality.