joukos/PaperTTY #91: Issue with Waveshare 5in83 v2

tahoewolfman, 2022-04-11:
The examples from waveshare work, but when using papertty it reports it's displaying but the only thing that happens is a black outline appears around the e ink screen nothing else. I am not a python coder... i did cross reference the waveshare files with the EPD5in83 driver but... it's all greek to me... any ideas? 

/home/zed/PaperTTY# papertty --driver epd5in83 terminal --portrait --noclear
Loading PIL font /home/zed/PaperTTY/papertty_venv/lib/python3.9/site-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
Started displaying /dev/vcsa1, minimum update interval 0.1 s, exit with Ctrl-C
Font /home/zed/PaperTTY/papertty_venv/lib/python3.9/site-packages/papertty/resources/tom-thumb.pbm doesn't support Unicode. Falling back to 8-bit encoding.

tahoewolfman, 2022-05-01:
Welp alrighty i'll just scrap this project and move on....

joukos, 2022-05-01:
Sorry for the late reply. At the moment there's no driver for the V2 and Waveshare says that:

> The controller and the driver codes are different and the codes of the two versions are not compatible with each other

So someone would need to write the driver first. As with most requests for device support, implementing the driver is best done with access to actual hardware. This unfortunately means expanding the driver support usually requires contributions from those who have these devices and can test the code with them - this particular model I don't happen to have. Writing a driver "blind" is of course possible, but I'm too tied up with other things at the moment to have spare time for it.

closed 2022-05-02 by tahoewolfman

tahoewolfman, 2022-05-02:
I hope you find time for yourself. Seriously seems like it's just you here. Sorry I came off rude. I Hit a deadline and had to start another project. While I don't like to leave a project unfinished, specially one that i've done before, its just how it goes. Worst comes to worst I can source a v1 eventually..