joukos/PaperTTY #98: Completely non functional with 2in13V2 and Pi Zero W

parkerlreed, 2023-01-06:
Latest Bullseye Lite

Both the C and Python examples from waveshare are working perfectly. 

Commands run but nothing printed on the screen.

```
parker@raspberrypi:~/eink/e-Paper/RaspberryPi_JetsonNano/python/examples $ sudo -E ~/.local/bin/papertty --driver EPD2in13v2 scrub 
Loading PIL font /home/parker/.local/lib/python3.9/site-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
parker@raspberrypi:~/eink/e-Paper/RaspberryPi_JetsonNano/python/examples $ echo Test | sudo -E ~/.local/bin/papertty --driver EPD2in13v2 stdin
Loading PIL font /home/parker/.local/lib/python3.9/site-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
```

It sits for about 10 seconds on the command doing "nothing"

Full strace 
[log.txt](https://github.com/joukos/PaperTTY/files/10356380/log.txt)

parkerlreed, 2023-01-06:
Tested a virtualenv in case the `local` installation was the cause but no luck there either.

joukos, 2023-01-14:
Does your board have some particular revision number? I've one that says Rev 2.1 on the silkscreen and it seems to have some issues with partial refresh (I forget if I had another unit that didn't have that problem, it's been a while), but works okay if I pass `--nopartial` after the driver selection. Although even without disabling partial refresh I do see things on the display but it's noisy.

parkerlreed, 2023-01-14:
It is indeed 2.1. I think I remember having to use nopartial forever ago but completely forgot about it haha. 

I'll give that a retest and let you know thank you.

Wil-Ro, 2023-04-20:
@parkerlreed did that end up fixing the problem?

parkerlreed, 2023-04-20:
Sorry, I've been unable to retest. I will close this out and reopen if nopartial doesn't work.

Thanks.

closed 2023-04-20 by parkerlreed