joukos/PaperTTY #99: Display doesn't update properly unless --invert

willemOH, 2023-01-14:
In both vnc and framebuffer (and I think the others but cannot confirm right now), my 3in7 waveshare has this behavior:

1. display color is inverted (or at least always white on black) without the --invert option
2. there is no full refresh (scrub?) after 50 updates (or after any amount specified with --fullevery) -> this may have to do with the fact that the scrub command does not work -> ValueError('Image must be the same dimensions as display required (280x480, got (16x480)
3. the display super ghosts to the point of filling up with garbage within 2 updates and being unreadable
4. using --invert with vnc fixes the super ghosting (its not even a ghost its just solid and not going away if that makes sense)
5. --invert does not invert everything (the i3bar in i3 stays black)

I've been through the edges of the issues reported here to get this far (vncdotool downgrade for one) and feel very close now! I just need to not have inversion of colors and perhaps a refresh (scrub? I haven't seen it in action so I don't know if its the same action) to get everything functional!! Very excited for that moment

joukos, 2023-01-14:
I don't have this model to test with and not sure if there's multiple revisions of it like some of the other boards do. But if yours seems to be inverted by default, might be worth trying to override the black/white values in the driver and seeing if that makes a difference, see here https://github.com/joukos/PaperTTY/blob/c2a5d72249e9a9484c9cc34245bb7a121891eb4b/papertty/drivers/drivers_base.py#L36-L38

The `--fullevery` option would just do a non-partial refresh, not scrubbing (which is another thing that should just be taken out, it's not very useful on well-behaved displays). While the 3.7" display supposedly supports partial refresh, the driver in PaperTTY does not implement that yet and every refresh should thus be a full one.

Adding partial refresh support shouldn't be too much work since Waveshare's own examples do it, but as I don't have the hardware to test with I haven't attempted it.

The 16x480 is a bit weird though...

willemOH, 2023-01-22:
Thanks for responding promptly. It's great to see an author keeping up with a project after some years. 

I took override to mean set my own value for black and white. I swapped them (white = 0, black = 255) to no effect. 

A new finding is that that `--invert` works as expected on GUI applications but for rxvt terminal emulator its always white on black inverted or not. Not sure what's up with that. May be a config thing for rxvt to more explicitly select terminal window colors idk. 

16x480 only happens with the `--scrub` command, no others. vnc resolution is 1280x720. 

The most glaring issue is that super ghosting I was talking about. It's like its not refreshing and only continuing to add to the e-ink screen. I made sure to put `--nopartial` since you said partial is not supported for 3in7. 

Though as mentioned, adding `--invert` solves this non-refreshing issue. It's just that everything is of course inverted. So I'd like to have it refreshing and not inverted. Perhaps there's a way to hack it so that using `--invert` goes through the function but sets the colors to just be normal? That was what I was attempting to do by switching the black and white values but to no avail.

Appreciate your support on this!

p.s does partial update increase response time on the display? If so I might have to send you a 3in7!