joukos/PaperTTY #101: 7in5_V2 unusable slow?

dwehrmann, 2023-03-20:
Hi there and thank you @joukos for pulling this off.

I know I am late to the party and I see others have successfully worked with the terminal and vnc here, so:

I have a pi zero wh and a waveshare 7.5 V2 connected, waveshare demos playing fine. I want to turn that into a simple typewriter.

Whenever I run the terminal command, things are taking like 20-30(!) seconds to refresh. I have seen others' videos out on yt and reddit running the exact same screen with acceptable speed. 

I tried the another of the screen model just to rule out a faulty device, connected it via the pinouts instead of the HAT, same results.
A Test with a 5.65 7-colors screen was not successful due to a drivers issue – but that's another story I suppose.

The command I'm running is:
`sudo papertty_venv/bin/papertty --driver EPD7in5V2 terminal --interactive`

I cannot install it via poetry and without using sudo it says something like "No read access to /dev/vcsa1 - maybe run with sudo?"
Is there anybody who can help me there? I would really like to get this going and it feels like I am very close ...

Thanks!

dadbean, 2023-04-15:
I got the 10.3 Hat version working with terminal and with x11vnc, but it is also painfully slow. Partial refresh works just fine, but it is very slow. Not 20seconds though, it is about 5.

dwehrmann, 2023-04-21:
Thanks. I assume it's a hardware issue, so this can be closed.

dwehrmann, 2023-04-21:
closing as stated above.

closed 2023-04-21 by dwehrmann