joukos/PaperTTY #102: No read access to /dev/vcsa1 - maybe run with sudo

alexmlucas, 2023-06-05:
Hi,
I get the message above after running the following command...
`poetry run papertty --driver epd5in83 terminal`
After adding `sudo` to the beginning of that command, terminal returns...
'sudo: poetry: command not found'
I expect this isn't an issue with paperTTY, more my lack of experience with Linux. However, any pointers would be greatly appreciated!

javierdepascual, 2023-09-11:
I've been trying to make papertty work today and thank god I figured out how to solve this issue. Apparently for sudo commands inside the virtual enviroment needs you to point to the specific directory (maybe I'm wrong but I saw this explained in another issue and it worked):

sudo papertty_venv/bin/papertty  --driver it8951 terminal

(replace it8951 with the driver you need from --list command)

joukos, 2023-09-13:
Thanks for sharing the tip here to help others. Hopefully some day I manage to update the docs from their woeful state.

closed 2023-09-13 by joukos