joukos/PaperTTY #119: Last known 'good' version of Raspbian OS for PaperTTY - and help in updating the install guide?

bodnerry, 2023-12-17:
Hi all, I'm a newcomer to the PaperTTY universe. 
I started trying to install PaperTTY on the current (15th Dec 2023) 64 bit Raspbian but soon got bogged down in errors. I'm assuming there are two ways to go:

1. install a version of Raspbian from the approximate date of the last update to PaperTTY
2. persevere with trying to address each complaint generated by the system, Poetry, pip, pipx, apt and the various other dramatis personae.

I imagine option 1. would be quicker (on a Pi3B) but that option 2. would be better for the future - to get it done and document it so there is a contemporary install guide for newcomers to work with.

I am happy to make nice documentation but my system knowledge is a bit too gappy to easily get PaperTTY up and running on a fresh install of Bookworm 64 bit, for example. 

Is anyone in a position to help me get something working?

I currently have a Pi3B and a Pi4(B?) and a Waveshare IT8951-based 6" ID display.

best wishes and thanks in advance,
Rodney

koroshiya, 2023-12-17:
I tried a fresh install of the latest 64 bit Raspberry Pi OS/Raspbian last month and it worked alright.
However, I installed via pip with a virtual environment, rather than the recommended poetry installation (which didn't work the last time I tried it).
From memory I think one of the dependencies might have been slightly different, eg. libtiff5-dev instead of libtiff5 or something like that. But otherwise the instructions for pip still worked on the latest raspbian. (I was using a 4b)

mcarr823, 2023-12-17:
I tried a fresh install of the latest 64 bit Raspberry Pi OS/Raspbian last month and it worked alright.
However, I installed via pip with a virtual environment, rather than the recommended poetry installation (which didn't work the last time I tried it).
From memory I think one of the dependencies might have been slightly different, eg. libtiff5-dev instead of libtiff5 or something like that. But otherwise the instructions for pip still worked on the latest raspbian. (I was using a 4b)

bodnerry, 2023-12-17:
thanks! I think libtiff5 was one of the things that didn't install. I'll try again with libtiff5-dev and see if there are knock-on effects. I tried installing with poetry. took a while to get pipx to install (used pip instead of apt in the end). 

To try th epip install, I'd type the following?:

```
python3 -m venv papertty_venv
source papertty_venv/bin/activate
pip install pappython3 ertty
papertty_venv/bin/papertty --help
```

mcarr823, 2023-12-17:
Yep, that should be it.
Enable SPI, install some dependencies with apt, then run those commands above to install via pip.

The first command creates a virtual python environment.
The second puts you in the virtual environment.
The third installs papertty in that environment.
(the fourth just runs the help command to show that it's working)

mcarr823, 2023-12-21:
@bodnerry I tried going through this again today. I guess something has changed since the build I tried last month, as I did need to install an extra dependency.

In addition to libtiff5-dev as mentioned above, I needed to install libfreetype-dev.
The tricky part is that libfreetype-dev needs to be installed before papertty.
If it's installed afterwards, then you need to reinstall pillow.
eg.
```
papertty_env/bin/pip3 uninstall pillow
sudo apt install libfreetype-dev
papertty_end/bin/pip3 install papertty
```

I also wound up creating a script to try and automate the entire PaperTTY setup from start to finish, on a fresh install of Raspberry Pi OS. So if you're still having trouble then that might be worth a try.

https://github.com/mcarr823/papertty-init

bodnerry, 2023-12-22:
Oooh! thanks for that. I'll give it a try this evening (I'm in Sydney) I
did get lots of errors when I tried installing with pip3.
I'll make a fresh SD card image. Just for info,

   - are you using a Pi3 or Pi4?
   - Which Raspbian OS are you using?
   - 32 or 64 bit?
   - I'm assuming the 64 bit OS is still fine with the Pi3? (I know I can
   probably google that one)

I'm excited because I never quite got papertty to work for me so I must've
missed some important info somewhere along the way that wasn't immediately
obvious.
I'm using the Waveshare 6" HD IT8951-based screen

Rod.

mcarr823, 2023-12-22:
I tried it with a pi4, 64bit OS, either Raspberry Pi OS or Raspberry Pi OS Lite (the script supports both).
Either 64bit or 32bit should work in theory, though I only tested with 64bit.

rpi3 should support either 64bit or 32bit.

bodnerry, 2023-12-24:
@mcarr823 I just fired up your script on a fresh full install of 64bit bookworm on a Pi3B (4gb I think?) and it all went well. I chose to try your fork. and most default options. Typing is still a bit laggy with a bluetooth keyboard but I'll find a USB one and see how that goes.

I'll try it again with 64bit Lite.

I was originally hoping to use a Pi Zero W and run on batteries but I suspect that will be a bit too slow for this, right? I got an under-voltage warning earlier with the Pi3B but now seems ok with a USB battery pack. I think maybe the socket was a bit loose before.

I imagine that the Pi4 and 5 will need a beefier battery set-up to run nicely. Most off-the-shelf phone-charger packs just boost 3.6v up to 5v and that is a bit shaky for a computer's power needs.

mcarr823, 2023-12-24:
When papertty starts up you should get a line of text show up along the lines of "lut_version = M841_TFA5210" or something similar. Could you see what yours is?

If it starts with M641, then you might be able to speed up your panel a bit further by enabling A2 mode. A2 mode should be a bit faster than the default DU mode (I think it's like 260ms vs 120ms).

A2 mode is enabled by default for all other IT8951 models. But the M641 (which some of the 6" panels use) is a bit different in how it works, so A2 mode is disabled by default.
If you edit papertty/drivers/driver_it8951.py there's a block of code like
```python3
            #This model requires four-byte alignment.
            #Don't enable a2 support until that has been implemented.
            #self.supports_a2 = True
```
If your model starts with M641, then you could try uncommenting that third line and restarting papertty to see if everything still works correctly (no visual artifacts and such).

A pi zero might work, but it depends on how fast you need it to be and where/how you're going to use this. If this is only going to be used at home, then one option would be to have the pi zero offload all the processing to a different device (eg. a desktop PC) and use the pi zero like a monitor. That's something which has only just been added and is currently pending as a pull request. I haven't written up instructions on that or added it to my fork or installer yet.

Otherwise the pi zero will probably be too slow. The 6" HD panel is 1448x1072, which is quite a lot of pixels for a pi zero to handle. The 3B should be a lot faster, especially if A2 works. A Pi Zero 2 WH might work, but I don't have one so I can't really say if it's fast enough.

I've been using a rpi 3A+ with Raspberry Pi OS Lite and a 1872x1404 panel. IIRC the delay caused by the pi's processing speed in terminal mode is around 60ms. For the 4b it was 8ms. For the 2b it was more like 200-250ms.

So if you add that time + the refresh rate latency (120ms for A2 mode, or 260ms for DU mode) + the keyboard latency (0 for a USB connection, something hopefully low for bluetooth) you should get a general idea of the speed.

A pi 4/5 will be faster, but I'd try USB and/or A2 mode first. The difference between a 3a+ and a 4b was only around 50ms in my testing, and the difference between the 4b and 5b can't be more than 8ms, so switching rpis seems like it would have diminishing returns.

bodnerry, 2023-12-25:
hmmm ... I changed to A2 in the driver.
I edited the one in: 

/home/pi/.local/share/papertty-init_venv/lib/python3.11/site-packages/papertty/drivers

 I'm using full raspbian rather than Lite, so maybe that is slowing things a little. Feels like about 250msecs between me hitting a key and the character appearing. I'm using a USB keyboard. The actual full screen refreshes feel quite snappy. Its just the delay when typing that feels off-putting.
[image]

bodnerry, 2023-12-25:
I'll make up another SD card with Lite to see if there is any difference.

bodnerry, 2023-12-25:
I forgot to say how much fun this is and how grateful I am for your help in getting this far. :)

mcarr823, 2023-12-25:
No problem. Here's hoping you get it running even smoother.

Based on your screenshot there are a couple of things you can try in order to speed it up:

- reduce font size (if it's not too small)
- reduce font width (eg. use a thinner font)
- disable word count (if you don't need it)
- use the rest of the screen. There might be a setting for this. Otherwise you can add a couple of empty lines (ie. press Enter a few times) then arrow up a line or two. So you'll always have an empty line or two below

Those first two mean the individual writes to the panel are smaller and thus faster. Same with image comparisons.
An example of a thinner font would be Intel One Mono Light: https://github.com/intel/intel-one-mono
You really want a Monospaced font for this. And a Light font should mean fewer pixels.

Disabling the word count does much the same thing, since you're not updating the text down the bottom of the screen. If less is changing on the screen then it's faster to draw.

Using the rest of the screen should be the big one. From your screenshot it's obvious you aren't getting a lot of text on the screen before it starts scrolling, and every time it scrolls you're redrawing half the screen. So anything which means you can put more text on the screen before that happens should help.

I would expect normal typing speed to be faster though. I know you said you enabled A2 mode (that is the right file - as long as you restarted papertty afterwards) but did you check the LUT version first? If it's something different than the ones listed in that source code file then it might not have helped.

You can also try setting sleep to a smaller value. eg. --sleep 0.01
It should only make a difference if you type slowly or pause often.

You could also set --vcom if the ribbon has a vcom value written on it. I don't know if it'll actually help the speed or not, but the vcom value written on the ribbon is what you should ideally run the driver at.

250ms doesn't seem unreasonable for a 3b though. I was getting around 60ms for the papertty logic with a 3a+. Add the A2 refresh (120ms) as well as the time taken to write to the pins and it's getting close to 200ms.

Do you have a 4b or 5b to try out? Comparing it with a faster pi would help you to determine whether the delay is because of the panel or because of the pi.

bodnerry, 2023-12-25:
Aren't I having a nerdy christmas?      :)
Having set up from your script with your fork, my Pi cheerfully boots straight into PaperTTY but doesn't display the LUT version on startup.

Is there a way to check the LUT version while PaperTTY is running?

My vcom sticker says **_-2.30_** so I'll tweak that in the driver.        

I have made a spare SD with Bookworm 64 Lite so I'll see if I get a bit more performance from that.

I have a Pi4 so I can compare that for speed too.

Wordgrinder appears to be built around the assumption that you will focus better if the text you are working on is always at the center of the screen (vertically) and that everything else should revolve around that.  The idea makes a lot of sense to me until, of course, we start trying it with EPD refresh rates.   I'll check in the Wordgrinder discussions and see what people are doing with that and whether Wordgrinder can be customised enough for e-ink happiness.

I find Nano quite easy to work with on the EPD screen because everything mostly stays put, and I could use markdown if I need to get fancy. I'm not likely to enjoy writing anything substantial in LaTEX though. :) I might even revisit VIM to see if my brain can be convinced that it's a good idea. then, of course there is emacs, which seems to be its own operating system!

Sorry, I'm maybe too chatty for an Issues discussion. I'll try to pin a _best practice with what we have so far_  summary (can we pin message on github?) so others can arrive and get what they need without reading the whole thing. :)

cheers and thanks again!

Rod

mcarr823, 2023-12-26:
> Having set up from your script with your fork, my Pi cheerfully boots straight into PaperTTY but doesn't display the LUT version on startup.
> 
> Is there a way to check the LUT version while PaperTTY is running?

That's weird. When you start papertty (with an IT8951 panel) it should print a few things to the command line.
In driver_it8951.py there are a few lines in the init function
```python3
        print("width = %d" % self.width)
        print("height = %d" % self.height)
        print("img_addr = %08x" % self.img_addr)
        print("firmware = %s" % firmware_version)
        print("lut = %s" % lut_version)
```
That last one should be the lut version.

I think nano is a good fit for this. The interface is minimal and the screen doesn't move until you hit the bottom.

I'm thinking I'll put the above performance tips (and other information) into a wiki on github at some point. Redoing the README is on my list lol.

bodnerry, 2023-12-26:
@mcarr823 with your script, I noticed a couple of warnings in the install. I chose gpiozero but it's still talking about RPi.GPIO - not a concern for me because it's running on a Pi3B but just checking. :)
------------------------------
Installing collected packages: spidev, RPi.GPIO, incremental, zope-interface, typing-extensions, six, pycryptodomex, Pillow, idna, constantly, click, attrs, hyperlink, automat, Twisted, vncdotool, papertty
  DEPRECATION: RPi.GPIO is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
  Running setup.py install for RPi.GPIO ... done
  DEPRECATION: Pillow is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
  Running setup.py install for Pillow ... done

bodnerry, 2023-12-26:
Haven't got the CLI version working yet. (Pi3B Raspbian lite latest bookworm).
I'd better stop obsessing and get some other things done and come back in a day or two. :)
The full Raspbian install running @mcarr823's fork in terminal mode works a treat though. the thinner font seemed to speed things up a little. 

The partial refresh seems to help but is there a keyboard shortcut to force a full refresh sometimes? I have a lot of missing text when I have to scroll about and edit or read back what I've written.

bodnerry, 2023-12-26:
when I run _./.local/bin/papertty-init/startpapertty.sh_ , I get the following:

> Traceback (most recent call last):
>   File "/home/pi/.local/share/papertty-init_venv/bin/papertty", line 5, in <module>
>     from papertty.papertty import cli
> ImportError: cannot import name 'cli' from 'papertty.papertty' (/home/pi/.local/share/papertty-init_venv/lib/python3.11/site-packages/papertty/papertty/__init__.py)

bodnerry, 2023-12-26:
This belongs elsewhere but, after asking the Wordgrinder author about scrolling behaviour in Wordgrinder, they said that the latest development version has a menu option under _Look and Feel_ to change to jump-scrolling so the page doesn't keep moving around you as you type at the center of the screen. Does not yet seem to be in the apt installed version for Pi.

In Nano, there are a few startup options that help with EPD displays. (You put them in your ~/.nanorc file): 
set jumpyscrolling    # Scroll the buffer contents per half-screen instead of per line.
set softwrap    # Display lines that exceed the screen’s width over multiple screen lines. 
set atblanks     # When soft line wrapping is enabled, make it wrap lines at blank characters (tabs and spaces) instead of always at the edge of the screen.

mcarr823, 2023-12-27:
> I chose gpiozero but it's still talking about RPi.GPIO

This happens when the script installs papertty via pip.
The script works by installing papertty the manual way (for its dependencies), then replaces papertty with a more up to date version from git.
The version of papertty in pip has RPi.GPIO as a dependency. So RPi.GPIO gets installed even if you select gpiozero.

> I noticed a couple of warnings in the install.

The warnings are from pip, because RPi.GPIO and Pillow 7.1.2 aren't in the new pip packaging format (I think).
RPi.GPIO hasn't been updated in a while, so it probably won't be updated to the new format.
For Pillow the solution is probably to target a newer version of Pillow (7.1.2 is quite old).
For RPi.GPIO the solution might end up being to either replace it with gpiozero or download RPi.GPIO manually.
I can work around this in the installer by installing the packages manually instead of by installing the pip version of papertty, but the better solution would be for papertty's pip version to be updated in one of the ways mentioned above, so the manual install method will continue to work.

> Enabling partial refresh seems to help too but is there a keyboard shortcut to force a full refresh sometimes? I have a lot of missing text when I have to scroll about and edit or read back what I've written.

There's no shortcut, though it should be simple enough to add a full refresh after x number of updates in terminal mode, just like VNC mode does.
That would be for dealing with ghosting though. I wouldn't expect for text to be missing.

> when I run _./.local/bin/papertty-init/startpapertty.sh_ , I get the following:

The error message suggests that papertty isn't installed in the venv.
Did the pip and git commands run successfully?
My guess would be that you don't have git installed. But the latest version of Raspberry Pi OS Lite should have git installed by default.
Wouldn't hurt to add it to the installer though.

atneik, 2024-02-18:
@spectre1989 looks like I'm running into the same issue using Raspberry Pi 5. It might be due to the change in GPIO hardware in the new Pi, so now RPi 5 can't use `RPi.GPIO` anymore. If that's the case, I think the project should move completely to `gpiozero` from `RPi.GPIO` and we should start a new issue.

```
(papertty_venv) aniket@raspberrypi:~/dev/PaperTTY $ sudo papertty_venv/bin/papertty --driver IT8951 terminal
gpiozero found - using that instead of RPi.GPIO
Loading PIL font /home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio'
  warnings.warn(
Traceback (most recent call last):
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 411, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=11, name='GPIO17', names=frozenset({17, 'BOARD11', '17', 'GPIO17', 'BCM17', 'J8:11', 'WPI0'}), pull='', row=6, col=1, interfaces=frozenset({'', 'uart', 'gpio', 'spi', 'dpi'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/papertty/drivers/drivers_base.py", line 121, in setup
    GPIO.pins[str(pin)] = OutputDevice(pin)
                          ^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/gpiozero/devices.py", line 553, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 413, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/gpiozero/pins/rpigpio.py", line 101, in __init__
    GPIO.setup(self._number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Cannot determine SOC peripheral base address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/aniket/dev/PaperTTY/papertty_venv/bin/papertty", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context().obj, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/papertty/papertty.py", line 1334, in terminal
    ptty = settings.get_init_tty()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/papertty/papertty.py", line 1059, in get_init_tty
    tty.init_display()
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/papertty/papertty.py", line 249, in init_display
    self.driver.init(partial=self.partial, vcom=self.vcom, enable_a2=self.enable_a2, enable_1bpp=self.enable_1bpp, mhz=self.mhz)
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/papertty/drivers/driver_it8951.py", line 181, in init
    if self.epd_init(includeDcPin=False) != 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/papertty/drivers/drivers_base.py", line 308, in epd_init
    GPIO.setup(self.RST_PIN, GPIO.OUT)
  File "/home/aniket/dev/PaperTTY/papertty_venv/lib/python3.11/site-packages/papertty/drivers/drivers_base.py", line 127, in setup
    rpiGPIO.setup(pin, rpiGPIO.OUT)
RuntimeError: Cannot determine SOC peripheral base address
```

mcarr823, 2024-02-22:
@atneik From your log
`gpiozero found - using that instead of RPi.GPIO`
it looks like gpiozero is installed and the device is trying to use it.
But the pin factory gpiozero is trying to use doesn't work with the pi5.

@spectre1989 I ran into the same issue.
I was able to get a little bit further by compiling the latest version of lgpio from git and copying that into the venv.
But then I ran into a separate issue of getting lgpio to actually work as the gpiozero backend.
It seems that lgpio wants to reserve the CS pin when setting up a SPI device.
It might be a matter of getting the SPI device to use a different select pin.
Or creating a class inherited from the lgpio spi device and exposing a method for interacting with the select pin.

spectre1989, 2024-02-24:
@atneik like @mcarr823 I had more luck compiling lgpio from scratch, pip only seems to have a really old version. Once I built it I converted the egg to a wheel and then installed it with pip.

The next issue I got around by going into drivers_base.py and commenting out:
`GPIO.setup(self.CS_PIN, GPIO.OUT)`
I got the idea from the waveshare examples, [this file](https://github.com/waveshareteam/e-Paper/blob/master/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py) specifically, this line:
`# self.GPIO_CS_PIN     = gpiozero.LED(self.CS_PIN)`

mcarr823, 2024-02-26:
@spectre1989 I've had another look at this and I was able to get lgpio working on a raspberry pi 4.
However, it required the use of Spidev instead of gpiozero for the SPI communication, then gpiozero for the pin operations.

The problem seems to be that the spidev kernel module reserves certain pins (notably the CS pin).
Rpi.GPIO is able to use the pin regardless, which is why the change worked on rpi4.
But gpiozero may not be able to (depending on the pin factory being used).
If the pin factory is lgpio, then it doesn't work, because lgpio doesn't like the fact that the spidev kernel module appears to be using the pin already, so lgpio won't reserve it and returns a Busy Pin error.
Whereas other factories in gpiozero, like pigpio, don't care if the pin appears to be in use.

The only gpiozero factory which currently works with the pi5 is the lgpio one.
Which means we're back to square one, because:
a) rpi.GPIO doesn't work on the pi5, so we can't use that for the pin changes, and
b) lgpio won't use the CS pin since the spidev kernel module appears to be using it.
So the above approach with the pi4 won't work for the 5.

gpiod (which also supports the pi5) seems to have the same problem, so changing libraries probably won't help.

I'm not sure how best to address this.
It might require a change to the lgpio source code to expose or manipulate the CS pin from the spi device.
Hopefully I'm missing something obvious, but I don't see any easy solution here.

In the case of the IT8951 board, I guess an alternative would be to use the USB driver instead of the GPIO pins.
I started work on that (there's a branch on my papertty fork), but currently it only supports 8bpp mode, and there's more work to be done.