joukos/PaperTTY #122: PaperTTY on Laptops with an IT8951 controller

zenmetsu, 2023-12-24:
I have a Lenovo Thinkbook Plus (Gen1) which has an IT8951 display which appears to be connected via USB as it shows up in `lsusb`.  I would be interested in trying to get this working.  It is possible that this device is connected via SPI as well, however I have not been successful in seeing the device over SPI.  Are there any plans to support the IT8951 via USB?

mcarr823, 2023-12-25:
I haven't tried running papertty over USB, but a quick look suggests that the commands are the same. So it might be possible to add a different type of target (USB instead of SPI) and translate everything as needed.

Some additional work would be needed in papertty itself to take advantage of the laptop's speed difference vs a raspberry pi. Terminal mode should already work well, but the other modes could be improved for this scenario.

In the meantime you could give this a go: https://github.com/blahgeek/rabbitink

That already runs over USB and specifically targets the IT8951 driver, so it might work for your laptop.

zenmetsu, 2023-12-25:
awesome!  Thanks for pointing me at that library.  I will take a stab at getting it working.  If not, I will try forking PaperTTY since it seems more full-featured.  I just wanted to avoid replication of effort in the event that someone else has already started a USB implementation.

zenmetsu, 2023-12-27:
The good news is that rabbitink will work on this machine, but only in gray mode and not in the faster mono A2 modes with dithering.  Since I know it is possible to drive this built-in display via USB in linux, I would be more than happy to help test any code for PaperTTY.  I'll leave this open until the repo-owner responds with information regarding potential USB support.

mcarr823, 2023-12-28:
@zenmetsu Fantastic news! I wonder what this means for other Lenovo laptops with e-ink screens?
The new Thinkbook Twist (Thinkbook Gen 4) where you can spin around the e-ink monitor would be of great interest (once they're significantly cheaper) since that one can be used as a legit e-ink laptop.
That one is a color e-ink screen, but the IT8951 driver should support 8 bits for color, so maybe it is using the same board?

Anyway, I'll have a go at adding USB support to PaperTTY in another couple of weeks, if nobody else does in the meantime.
I'll comment on this issue when I've got something. I should be able to test the USB part, but if you would be willing to confirm whether it works on your laptop once it's done that would be greatly appreciated.

zenmetsu, 2023-12-28:
the rabbitink dev was sending the regular commands to switch to 1bpp, but my display apparently did not accept the command.  i don't think it is a controller issue, since i'm only aware of a single IT8951 spec, but it is possible that the eink panel needs something special, or maybe it just doesn't support 1bpp.  The workaround was to send dithered monochrome 8bpp and forego greyscale for the speed mode.  This worked surprisingly well and obviated the need for disruptive screen refreshes.

I'll be happy to test your code out, but I suspect a similar issue may be encountered at first.

mcarr823, 2023-12-29:
Yeah, I had a quick look through the issue on the rabbitink issue you opened. It'll be interesting to get some more info from the panel, like the LUT version, which papertty prints from the IT8951 panel. That could potentially have something to do with the 1bpp issue.
Or it could be to do with the image height. IIRC rabbitink forces 4-byte alignment for the x axis and width, but doesn't have anything similar for the y axis or height. I ran into some issues during testing with the IT8951 panel when the image height wasn't a multiple of 16.
It'll also be worth trying DU mode if A2 doesn't work out (I think rabbitink uses either A2 or GC16, depending on the number of colors). Or 2bpp mode if 1bpp doesn't work out.

...But all of that will have to wait until papertty actually supports USB data transfer lol.

zenmetsu, 2023-12-29:
I would be happy to send you whatever info that I can.  Just tell me which commands you would like me to run.

For what it is worth, I thought I saw something about different A2 modes... like i recall a 4 and a 6 submode, where the larger panels used 6.  

I am not in any rush on the eink display since it is of limited use to me since i cannot get stylus or touch input to work.  xsetwacom sees the devices, but i can't get any input to show up.  :/

mcarr823, 2023-12-30:
Yeah, 6 is for A2 mode on most panels, 4 is for A2 mode on some of the 6" panels. I figured maybe the lenovo one uses something different than the default 6.

The LUT version is part of the output when starting up the IT8951 driver with papertty. So once we've added USB support it should be output to the command line on startup.

mcarr823, 2024-01-10:
Quick update on this.

I've got the driver more or less working and a test script to run it.
Only 8bpp right now. Also, text seems to work, but images weren't working the last time I tested.

The driver and test script are here: https://github.com/mcarr823/PaperTTY/tree/usb-it8951
I'm still working on integrating the driver with papertty itself, so that hasn't been committed yet.
Code needs commenting/cleaning up, but that'll be after it actually works lol.