joukos/PaperTTY #128: Adjusting PaperTTY to work with weAct 2.9

Uri0533, 2024-02-02:
Hi guys, first of all, thank you very much for your amazing work!

I am happy to announce that the weAct 2.13  E-Paper display works successfully with the EPD2in13v4 driver from @mcarr823's fork.

The weAct 2.9 E-Paper display is compatible with the Waveshare 2in9v2 screen - meaning it works perfectly with the Waveshare python sample. Since PaperTTY doesn't support Waveshare 2in9v2 at the moment, I tried to see which of the other drivers it works well with.
I was happy to find that this display also works great with the EPD2in13v4 driver, except for one problem - a large part of the screen on the left side does not work - that is, it remains the same color as before. (In the picture below you can see part of the Waveshare logo from the example) 
I tried to edit the driver EPD2in13v4 in drivers_partial.py, and increase its width (for some reason it is marked as height...) from 250 to 296, but the result was that now the text seems to be **behind The inactive area**. When I tried to edit other parts of the code, the code just crashed, I'm not a very good python programmer...

I would appreciate guidance on which sections of the code need to be modified so that the entire screen will be active.
Or guidance on how to add the Waveshare 2in9v2 driver to PaperTTY.

[image]

mcarr823, 2024-02-08:
I don't have an epd2in9v2, so this is currently untested, but I took a stab at porting the waveshare driver over.
You can try it out here: https://github.com/mcarr823/PaperTTY/tree/epd2in9v2

Uri0533, 2024-02-08:
Thank you very much @mcarr823, for your help.
With the new EPD2in9v2 driver you created the display is now fully
functional!

However, the refresh rate of the  EPD2in9v2 driver is slower than the
refresh rate of the  EPD2in13v4 .
Do you have any idea what is the reason for this, and can it be improved?

Many thanks!!!

‫בתאריך יום ה׳, 8 בפבר׳ 2024 ב-7:07 מאת ‪mcarr823‬‏ <‪
***@***.***‬‏>:‬

mcarr823, 2024-02-08:
@Uri0533 I think it probably can be made to run faster, but it's hard to say without a device to test with.
Right now the code is written more or less the same as the code from Waveshare with only minor modifications.
To potentially make it faster I'd start by tinkering with the display_partial function.
eg. Seeing if it works if the LUT is loaded only once instead of every time, or seeing if set_memory_area and set_memory_pointer can be changed to allow for partial writes.

mcarr823, 2024-02-09:
Actually, it might just be the panel.
https://www.waveshare.com/wiki/Pico-ePaper-2.13
https://www.waveshare.com/wiki/Pico-ePaper-2.9
The 2.13 is listed as Partial refresh time: 0.3s
The 2.9 is listed as Partial refresh time: 0.6s

Uri0533, 2024-02-09:
@mcarr823, I doubt it's the panel.
With the same panel - weAct2.9 -  when I run the  EPD2in13v4, it has one
speed, and when I run the  EPD2in9v2  it has a different, slower speed.

An ultimate solution would be to change EPD2in13v4 so it will recognize the
entire display...

Thank you very much for your help.

> Message ID: ***@***.***>
>

mcarr823, 2024-02-13:
Interesting that the code for the epd2in13v4 works on the epd2in9v2 at all.
I guess the code is fairly similar. The main difference is the LUT loading.
You could try commenting out the calls to set_lut and see if it makes a difference.