joukos/PaperTTY #133: Raspberry pi 5 support

BTomiii, 2024-09-15:
I have a raspberry pi 5, but unfortunately this isn't working. The official waveshare wiki states, that the raspberry pi 5 need to use a different GPIO library (namely: Install lg library (Recommended for Pi5, BCM library currently does not support Pi5)). Could anyone provide support for this. Or maybe help on what I need to modify to work with it?
Here is the error message i get:

$ papertty_venv/bin/papertty --driver IT8951 scrub
Loading PIL font /home/pi/papertty_venv/lib/python3.11/site-packages/papertty/resources/tom-thumb.pil. Font size is ignored.
Traceback (most recent call last):
  File "/home/pi/papertty_venv/bin/papertty", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/home/pi/papertty_venv/lib/python3.11/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/papertty_venv/lib/python3.11/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/pi/papertty_venv/lib/python3.11/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/papertty_venv/lib/python3.11/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/papertty_venv/lib/python3.11/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/papertty_venv/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context().obj, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/papertty_venv/lib/python3.11/site-packages/papertty/papertty.py", line 579, in scrub
    ptty = settings.get_init_tty()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/papertty_venv/lib/python3.11/site-packages/papertty/papertty.py", line 462, in get_init_tty
    tty.init_display()
  File "/home/pi/papertty_venv/lib/python3.11/site-packages/papertty/papertty.py", line 251, in init_display
    self.driver.init(partial=self.partial)
  File "/home/pi/papertty_venv/lib/python3.11/site-packages/papertty/drivers/driver_it8951.py", line 179, in init
    GPIO.setup(self.RST_PIN, GPIO.OUT)
RuntimeError: Cannot determine SOC peripheral base address

BTomiii, 2024-09-23:
I found a very useful piece of software: [GPIOconverter](https://github.com/bobrathbone/GPIOconverter)
Using this, and rewriting a few lines in drivers_base.py I was able to get it working on pi 5.

Vansick1, 2024-10-11:
@BTomiii , can you post your modified code in this thread for future reference? There's apparently a conga-line of developers having similar issues with GPIO on the Pi5

BTomiii, 2024-10-11:
I made a branch and pull request already. You can follow the instructions there.

Vansick1, 2024-10-11:
@BTomiii Thank you so much!

BTomiii, 2024-10-11:
@Vansick1 , let me know if you got it working, or want any help.