joukos/PaperTTY #139: poetry.lock

parrelld, 2025-03-31:
Hello, this is my first git hub post, please let me know if I can provide you additional information. I am installing papery on a raspberry pi zero 2W. I am using the latest 64-bit raspianOS lite Kernel version: 6.6, Debian version: 12 (bookworm), from 11-19-2024. Python is version "Python 3.11.2" and papertty is version "papertty 0.1.8".  I have installed part of lightdm desktop manager. If you think any of that is an issue I can provide more detail. When following the directions on GitHub I run into a few issues that I have not resolved yet. When I get to the step to run "poetry update" I get the following error.

`The lock file is not compatible with the current version of Poetry.
Regenerate the lock file with the "poetry lock" command.`

Running "poetry lock" gives the same exact error. At this point I did a little searching and the second comment under the solution at this [link](https://stackoverflow.com/questions/62040724/warning-the-lock-file-is-not-up-to-date-with-the-latest-changes-in-pyproject-to) suggested removing the poetry.lock file. So I did, I ran the following.

```
$ rm poetry.lock 
$ poetry lock
Updating dependencies
Resolving dependencies... (35.2s)

Writing lock file
```

Then I ran "poetry install" again. It was a different result but I still got an error. The output is long.

```
$ poetry install
Installing dependencies from lock file

Package operations: 5 installs, 0 updates, 0 removals

  - Installing twisted (20.3.0): Failed

PEP517 build of a dependency failed

Backend subprocess exited when trying to invoke build_wheel
```
**_**NOTE from poster: The next part is a ton of dialog that I can share if you want to see but I believe is just the result of this command " | Command '['/tmp/tmpp5zyocyn/.venv/bin/python', '/home/ewriter/.local/pipx/venvs/poetry/lib/python3.11/site-packages/pyproject_hooks/_in_process/_in_process.py', 'build_wheel', '/tmp/tmpnh5ko2ug']' returned non-zero exit status 1.
" 
The dialog continues after this. I tried to include what I think triggered the error**_**
```
 | aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/tmp/tmpp5zyocyn/.venv/include -I/usr/include/python3.11 -c src/twisted/test/raiser.c -o build/temp.linux-aarch64-cpython-311/src/twisted/test/raiser.o
    | src/twisted/test/raiser.c:198:12: fatal error: longintrepr.h: No such file or directory
    |   198 |   #include "longintrepr.h"
    |       |            ^~~~~~~~~~~~~~~
    | compilation terminated.
    | error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1

Note: This error originates from the build backend, and is likely not a problem with poetry but one of the following issues with twisted (20.3.0)

  - not supporting PEP 517 builds
  - not specifying PEP 517 build requirements correctly
  - the build requirements are incompatible with your operating system or Python version
  - the build requirements are missing system dependencies (eg: compilers, libraries, headers).

You can verify this by running pip wheel --no-cache-dir --use-pep517 "twisted (==20.3.0)".
```

I tried its suggestion to run the command above. Here is the command and its output below. The output it really long. I am going to just give you what I think are the important bits' please let me know if you want to see more.

```
$ pip wheel --no-cache-dir --use-pep517 "twisted (==20.3.0)"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting twisted==20.3.0
  Downloading Twisted-20.3.0.tar.bz2 (3.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.1/3.1 MB 3.3 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting zope.interface>=4.4.2
  Downloading zope.interface-7.2-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (260 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 261.0/261.0 kB 3.5 MB/s eta 0:00:00
Collecting constantly>=15.1
  Downloading https://www.piwheels.org/simple/constantly/constantly-23.10.4-py3-none-any.whl (13 kB)
Collecting incremental>=16.10.1
  Downloading https://www.piwheels.org/simple/incremental/incremental-24.7.2-py3-none-any.whl (20 kB)
[... 204 lines cut ...]
      copying src/twisted/internet/test/fake_CAs/not-a-certificate -> build/lib.linux-aarch64-cpython-311/twisted/internet/test/fake_CAs
      copying src/twisted/internet/test/fake_CAs/thing1.pem -> build/lib.linux-aarch64-cpython-311/twisted/internet/test/fake_CAs
      copying src/twisted/internet/test/fake_CAs/thing2-duplicate.pem -> build/lib.linux-aarch64-cpython-311/twisted/internet/test/fake_CAs
      copying src/twisted/internet/test/fake_CAs/thing2.pem -> build/lib.linux-aarch64-cpython-311/twisted/internet/test/fake_CAs
      copying src/twisted/internet/iocpreactor/notes.txt -> build/lib.linux-aarch64-cpython-311/twisted/internet/iocpreactor
      copying src/twisted/trial/test/_assertiontests.py.3only -> build/lib.linux-aarch64-cpython-311/twisted/trial/test
      running build_ext
      building 'twisted.test.raiser' extension
      creating build/temp.linux-aarch64-cpython-311/src/twisted/test
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/include/python3.11 -c src/twisted/test/raiser.c -o build/temp.linux-aarch64-cpython-311/src/twisted/test/raiser.o
      src/twisted/test/raiser.c:198:12: fatal error: longintrepr.h: No such file or directory
        198 |   #include "longintrepr.h"
            |            ^~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for twisted
Failed to build twisted
ERROR: Failed to build one or more wheels

```

So this is where I have paused. Obviously some of this is outside the suggested steps, so I am happy to abandon it, just thought I would show what I found and tried. I also tried running update and upgrade with apt-get, and "poetry update" followed by the above workflow again after completely removing the papery directory and recloning, and I had the same results. I am not in a big rush, but please let me know if I can help more. Thank you!

parrelld, 2025-04-07:
I am going to try with a fresh install I am not convinced my other playing arounds didn't get in the way.

mountwebs, 2025-04-08:
Take a look here: https://github.com/joukos/PaperTTY/issues/119#issuecomment-1859075089

I gave up using Poetry and installed with pip to virtual env instead. That worked fine for me.

mountwebs, 2025-04-08:
Btw. Please share if you get Poetry to work :)

parrelld, 2025-04-08:
Thank you! That thread looks helpful! I will have a bit of time on the weekend to give it a go. I'll try poetry a bit more, but I am glad to know pip is a good alternative. I'll report back.

I will reshare here a[ link](https://stackoverflow.com/questions/62040724/warning-the-lock-file-is-not-up-to-date-with-the-latest-changes-in-pyproject-to) I posted above and distill where I am at with poetry for you since my first post was a lot all at once. There is some guidance in that thread about the poetry.lock file and versions of poetry past 2.0.0. Running "poetry lock" did not help me. Deleting poetry.lock (I actually just renamed it to back it up) allowed me to get further and install some dependencies with poetry. I get stuck with "twisted" but I haven't had the time to dive into it. It suggests a command (see code below) that I tried and it runs, but finishes with error. I think twisted is the main issue right now with this workflow. I have to confess a lack of knowledge about these dependencies currently though.
`pip wheel --no-cache-dir --use-pep517 "twisted (==20.3.0)"`
I did try this above, it is the last two big chunks of "code". Thanks for the suggestions!